platformio/platformio-home #640: Could not initialize project

retsel89, 2020-04-22:
PIO Core Call Error: "The current working directory /Users/lester/Documents/PlatformIO/Projects/CAN_IDF_Framework will be used for the project.\n\nThe next files/directories have been created in /Users/lester/Documents/PlatformIO/Projects/CAN_IDF_Framework\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html\nPLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40000.200303 (4.0.0) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \n - toolchain-xtensa32 2.80200.200226 (8.2.0)\nInstalling ESP-IDF's Python dependencies\nCollecting cryptography>=2.1.4\n  Downloading cryptography-2.9.1-cp35-abi3-macosx_10_9_x86_64.whl (1.6 MB)\nCollecting future>=0.15.2\n  Downloading future-0.18.2.tar.gz (829 kB)\nCollecting pyparsing<2.4.0,>=2.0.3\n  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)\nCollecting six>=1.4.1\n  Downloading six-1.14.0-py2.py3-none-any.whl (10 kB)\nCollecting cffi!=1.11.3,>=1.8\n  Downloading cffi-1.14.0-cp37-cp37m-macosx_10_9_x86_64.whl (174 kB)\nCollecting pycparser\n  Downloading pycparser-2.20-py2.py3-none-any.whl (112 kB)\nBuilding wheels for collected packages: future\n  Building wheel for future (setup.py): started\n  Building wheel for future (setup.py): finished with status 'done'\n  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=a14da20a071e3f4f0adb523cd8608aad95ebbfdcd5dbcc00b8b8b9396a8c6e65\n  Stored in directory: /Users/lester/Library/Caches/pip/wheels/56/b0/fe/4410d17b32f1f0c3cf54cdfb2bc04d7b4b8f4ae377e2229ba0\nSuccessfully built future\nInstalling collected packages: six, pycparser, cffi, cryptography, future, pyparsing\nERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/Users/lester/.platformio/penv/lib/python3.7/site-packages/six.py'\nConsider using the `--user` option or check the permissions.\n\n*** Error 1\nReading CMake configuration...\n-- Found Git: /usr/bin/git (found version \"2.20.1 (Apple Git-117)\") \n-- IDF_TARGET not set, using default target: esp32\n-- The C compiler identification is GNU 8.2.0\n-- The CXX compiler identification is GNU 8.2.0\n-- The ASM compiler identification is GNU\n-- Found assembler: /Users/lester/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/lester/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/lester/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc -- works\n-- Detecting C compiler ABI info\n-- Detecting C compiler ABI info - done\n-- Detecting C compile features\n-- Detecting C compile features - done\n-- Check for working CXX compiler: /Users/lester/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++\n-- Check for working CXX compiler: /Users/lester/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++ -- works\n-- Detecting CXX compiler ABI info\n-- Detecting CXX compiler ABI info - done\n-- Detecting CXX compile features\n-- Detecting CXX compile features - done\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\n-- Project version: 1\n-- Building ESP-IDF components for target esp32\n-- Checking Python dependencies...\nThe following Python requirements are not satisfied:\nfuture>=0.15.2\ncryptography>=2.1.4\npyparsing>=2.0.3,<2.4.0\nPlease refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.\nAlternatively, you can run \"/Users/lester/.platformio/penv/bin/python -m pip install -r /Users/lester/.platformio/packages/framework-espidf/requirements.txt\" for resolving the issue.\n-- Configuring incomplete, errors occurred!\nSee also \"/Users/lester/Documents/PlatformIO/Projects/CAN_IDF_Framework/.pio/build/esp32doit-devkit-v1/CMakeFiles/CMakeOutput.log\".\n\nfatal: not a git repository (or any of the parent directories): .git\nCMake Error at /Users/lester/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):\n  Some Python dependencies must be installed.  Check above message for\n  details.\nCall Stack (most recent call first):\n  /Users/lester/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)\n  /Users/lester/.platformio/packages/framework-espidf/tools/cmake/project.cmake:337 (idf_build_process)\n  CMakeLists.txt:3 (project)\n\n\n\n========================= [FAILED] Took 10.36 seconds ========================="

valeros, 2020-04-22:
Hi @retsel89 ! Strange, it looks like you don't have required permissions:
```
ERROR: Could not install packages due to an Environment
Error: [Errno 13] Permission denied: '/Users/lester/.platformio/penv/lib/python3.7/site packages/six.py
```

retsel89, 2020-04-22:
Hi

Going to check that, thanks

closed 2020-09-17 by stale[bot]