platformio/platformio-home #650: Could not list directory /home/stephane

stephdaniel133, 2020-04-25:
'utf-8' codec can't encode character '\udc90' in position 3797: surrogates not allowed

valeros, 2020-05-21:
Please provide an output from the system terminal:
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```
To me it looks like is caused by Python 2.x. Could you please try to install Python 3.7
https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

stephdaniel133, 2020-05-22:
Hello,
To be precise, I'm on linux OpenSuse 15.1.
The following commands have been executed in the Vscode terminal.

Echo $PATH:
/home/stephane/bin:/usr/local/bin:/usr/bin:/bin:/snap/bin

Echo $PYTHONPATH: (print a empty line)

python --version
Python 2.7.14

pip --version
pip 20.1.1 from /home/stephane/.platformio/penv/lib64/python3.6/site-packages/pip (python 3.6)

pip list
Package          Version
---------------- ----------
bottle           0.12.18
certifi          2020.4.5.1
chardet          3.0.4
click            7.1.2
colorama         0.4.3
idna             2.9
marshmallow      3.6.0
pip              20.1.1
platformio       4.3.3
pyelftools       0.26
pyserial         3.4
requests         2.23.0
semantic-version 2.8.5
setuptools       40.6.2
tabulate         0.8.7
urllib3          1.25.9
wheel            0.34.2

On the default configuration of OpenSuse 15.1, their is python 3.6.10 installed. I try to install python 3.7.2 but I don't understand why it doesn't work.

Thanks

closed 2020-09-16 by stale[bot]