platformio/platformio-home #662: Could not initialize project

WrongBit, 2020-04-28:
PIO Core Call Error: "The current working directory\n C:\\Users\\thorn\\Documents\\PlatformIO\\Projects\\BlinkTheLED \nwill be used for the project.\n\nThe next files/directories have been created in \u001b[36mC:\\Users\\thorn\\Documents\\PlatformIO\\Projects\\BlinkTheLED\u001b[0m\n\u001b[36minclude\u001b[0m - Put project header files here\n\u001b[36mlib\u001b[0m - Put here project specific (private) libraries\n\u001b[36msrc\u001b[0m - Put project source files here\n\u001b[36mplatformio.ini\u001b[0m - Project Configuration File\nPlatformManager: Installing \u001b[36matmelavr\u001b[0m\nDownloading...\nUnpacking...\natmelavr @ 2.0.0 has been successfully installed!\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\thorn\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\thorn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\thorn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\thorn\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\thorn\\.platformi

valeros, 2020-04-30:
Looks like you didn't attach the entire error message so it's hard to guess what happened.

WrongBit, 2020-05-01:
hi, @valeros! It's all what copied from error box, I didn't copy it manually.
But looks like sh%%t like Python is NOT suitable even for medium-level projects. You have line:

    Error: Traceback (most recent call last):\n File "c:\users\thorn\.platformio\penv\lib\site-packages\platformio\main.py", line 102, in main\n cli() # pylint: disable=no-value-for-parameter\n File "c:\users\thorn\.platformio\penv\lib\site-packages\click\core.py", line 829, in call\n return self.main(*args, **kwargs)\n File "c:\users\thorn\.platformio\penv\lib\site-packages\click\core.py", line 782

...and still cannot say anything. It's pity, on C# I could be more helpful.

closed 2020-09-17 by stale[bot]