platformio/platformio-home #671: Could not initialize project

VuoriAndrey, 2020-04-29:
PIO Core Call Error: "The current working directory C:\\Users\\Андрей\\Documents\\PlatformIO\\Projects\\wifi will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Андрей\\Documents\\PlatformIO\\Projects\\wifi\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\r\nPLATFORM: Espressif 32 1.12.1 > Espressif ESP32 Dev Module\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40000.200303 (4.0.0) \r\n - tool-cmake 3.16.5 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nReading CMake configuration...\r\n-- Could NOT find Git (missing: GIT_EXEC

VuoriAndrey, 2020-04-29:
may be  "Андрей"

valeros, 2020-04-30:
Hi @VuoriAndrey ! Strange issue, do you have `Git` installed in your sysytem?

VuoriAndrey, 2020-05-01:
Hi Valerii!
Yes. I have.

valeros, 2020-05-01:
Is the path to Git executable added to PATH environment variable? CMake complains that Git is missing:
```
 Could NOT find Git (missing: GIT_EXEC
```

closed 2020-08-31 by valeros