platformio/platformio-home #765: Could not initialize project

buddikatdk, 2020-05-12:
PIO Core Call Error: "The current working directory\n C:\\Users\\USER\\Documents\\PlatformIO\\Projects\\pro1 \nwill be used for the project.\n\nThe next files/directories have been created in \u001b[36mC:\\Users\\USER\\Documents\\PlatformIO\\Projects\\pro1\u001b[0m\n\u001b[36minclude\u001b[0m - Put project header files here\n\u001b[36mlib\u001b[0m - Put here project specific (private) libraries\n\u001b[36msrc\u001b[0m - Put project source files here\n\u001b[36mplatformio.ini\u001b[0m - Project Configuration File\nPlatformManager: Installing \u001b[36matmelavr\u001b[0m\nDownloading...\nUnpacking...\natmelavr @ 2.1.0 has been successfully installed!\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 138, in project_init\n    ctx, project_dir, board, project_option, env_prefix, ide is not None\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 434, in fill_project_envs\n    _install_dependent_platforms(ctx, used_platforms)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 445, in _install_dependent_platforms\n    cli_platform_install, platforms=list(set(platforms) - set(installed_platforms))\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 325, in platform_install\n    force=force,\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 103, in install\n    force=force,\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 307, in install_packages\n    self.pm.install(name, version, silent=silent, force=force)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 686, in install\n    self.print_message(msg)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 608, in print_message\n    click.echo(\"%s: %s\" % (self.__class__.__name__, message), nl=nl)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo\n    \"%s\\n\" % (args[0] if args else \"\")\n  File \"c:\\users\\user\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n    value.decode() if is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

buddikatdk, 2020-05-12:
above issue was obtained when opening arduino project with vscode please help

buddikatdk, 2020-05-12:
command 'multiclip.cut' not found in vscode

valeros, 2020-05-12:
Hi @buddikatdk ! What project did you try to open?

closed 2020-06-29 by buddikatdk