platformio/platformio-home #766: Problem installing libraries that I aread used on Arduino IDE

morallito, 2020-05-12:
I'm trying to convert a project from Arduino IDE to PIO. But when I try to install some libraries thar I've aready use I'm gerring the following error. When I try to instal opther libs that I've never used on Arduino IDE everithing goes well. 

```PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\Iagoh\\.platformio\\lib\u001b[0m\nLibraryManager: Installing \u001b[36mid=31\u001b[0m @ ^1.1.2\nAdafruit Unified Sensor @ 1.1.2 is already installed\nInstalling dependencies\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 155, in lib_install\n    library, silent=silent, interactive=interactive, force=force\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 320, in install\n    lib_id = self.search_lib_id(filters, silent, interactive)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 149, in search_lib_id\n    % click.style(filters[\"name\"], fg=\"cyan\")\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo\n    \"%s\\n\" % (args[0] if args else \"\")\n  File \"c:\\users\\iagoh\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n    value.decode() if is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

```

valeros, 2020-05-12:
Hi @morallito ! Could you please provide a full list of used libraries?

morallito, 2020-05-14:
The libs that i've been using was: 

[Adafruit sensor](https://github.com/adafruit/Adafruit_Sensor)
[Adafruit bno055](https://github.com/adafruit/Adafruit_Python_BNO055)

I've solved the problem downloading it and pasting direcly on the libs folder.

valeros, 2020-05-14:
Thanks, for the info. BTW, do you have any non-ascii symbols in the path to your project? Maybe username?

morallito, 2020-07-22:
Yeah, I've used some Brazilian characters on the instalation folder "ç". I've reinstalled everything and now it's working. Thanks

closed 2020-07-22 by morallito