platformio/platformio-home #773: Could not initialize project

AlucardMode0, 2020-05-14:
PIO Core Call Error: "The current working directory C:\\Users\\dante\\Documents\\PlatformIO\\Projects\\test will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\dante\\Documents\\PlatformIO\\Projects\\test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing xmc4700_relax_kit (platform: infineonxmc; board: xmc4700_relax_kit; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/infineonxmc/xmc4700_relax_kit.html\r\nPLATFORM: Infineon XMC 1.1.4 > XMC4700 Relax Kit\r\nHARDWARE: XMC4700 144MHz, 1.95MB RAM, 2.00MB Flash\r\nDEBUG: Current (jlink) On-board (jlink)\r\nPACKAGES: \r\n - framework-arduinoxmc 1.4.0 \r\n - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nUnicodeDecodeError: 'utf-8' codec can't decode byte 0x92 in position 257: invalid start byte:\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 156:\r\n    env.SConscript(\"$BUILD_SCRIPT\")\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 597:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 286:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\dante\\.platformio\\platforms\\infineonxmc\\builder\\main.py\", line 142:\r\n    target_elf = env.BuildProgram()\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 62:\r\n    env.ProcessProjectDeps()\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 131:\r\n    project_lib_builder = env.ConfigureProjectLibBuilder()\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 1038:\r\n    lib_builders = env.GetLibBuilders()\r\n  File \"C:\\Users\\dante\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 963:\r\n    lb = LibBuilderFactory.new(env, lib_dir)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 59:\r\n    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 118:\r\n    self._manifest = manifest if manifest else self.load_manifest()\r\n  File \"C:\\Users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 470:\r\n    return ManifestParserFactory.new_from_file(manifest_path).as_dict()\r\n  File \"c:\\users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manifest\\parser.py\", line 74:\r\n    cls.read_manifest_contents(path), type_from_uri, remote_url\r\n  File \"c:\\users\\dante\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manifest\\parser.py\", line 64:\r\n    return fp.read()\r\n  File \"C:\\Users\\dante\\AppData\\Local\\Programs\\Python\\Python36-32\\lib\\codecs.py\", line 321:\r\n    (result, consumed) = self._buffer_decode(data, self.errors, final)\r\n========================== [FAILED] Took 1.45 seconds =========================="

AlucardMode0, 2020-05-14:
what do i need to do ?

valeros, 2020-05-21:
Please provide an output from the system terminal:
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```
To me it looks like is caused by Python 2.x. Could you please try to install Python 3.7
https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

valeros, 2020-05-29:
Duplicate https://github.com/Infineon/platformio-infineonxmc/issues/12

closed 2020-05-29 by valeros