platformio/platformio-home #775: Could not initialize project

k0d3x8its, 2020-05-14:
PIO Core Call Error: "toolchain-gccarmnoneeabi @ 1.80201.181220 has been successfully installed!\nPackageManager: Installing framework-zephyr @ ~2.20200.0\nframework-zephyr @ 2.20200.200401 has been successfully installed!\nPackageManager: Installing framework-zephyr-hal-nordic @ ~0.20200.200130\nframework-zephyr-hal-nordic @ 0.20200.200130 has been successfully installed!\nPackageManager: Installing framework-zephyr-civetweb @ ~0.20200.190807\nframework-zephyr-civetweb @ 0.20200.190807 has been successfully installed!\nPackageManager: Installing framework-zephyr-fatfs @ ~0.20200.191125\nframework-zephyr-fatfs @ 0.20200.191125 has been successfully installed!\nPackageManager: Installing framework-zephyr-libmetal @ ~0.20200.190530\nframework-zephyr-libmetal @ 0.20200.190530 has been successfully installed!\nPackageManager: Installing framework-zephyr-lvgl @ ~0.20200.191207\nframework-zephyr-lvgl @ 0.20200.191207 has been successfully installed!\nPackageManager: Installing framework-zephyr-mbedtls @ ~0.20200.200206\nframework-zephyr-mbedtls @ 0.20200.200206 has been successfully installed!\nPackageManager: Installing framework-zephyr-mcumgr @ ~0.20200.200117\nframework-zephyr-mcumgr @ 0.20200.200117 has been successfully installed!\nPackageManager: Installing framework-zephyr-open-amp @ ~0.20200.190612\nframework-zephyr-open-amp @ 0.20200.190612 has been successfully installed!\nPackageManager: Installing framework-zephyr-openthread @ ~0.20200.191014\nframework-zephyr-openthread @ 0.20200.191014 has been successfully installed!\nPackageManager: Installing framework-zephyr-segger @ ~0.20200.190421\nframework-zephyr-segger @ 0.20200.190421 has been successfully installed!\nPackageManager: Installing framework-zephyr-tinycbor @ ~0.20200.191016\nframework-zephyr-tinycbor @ 0.20200.191016 has been successfully installed!\nPackageManager: Installing framework-zephyr-littlefs @ ~0.20200.190811\nframework-zephyr-littlefs @ 0.20200.190811 has been successfully installed!\nPackageManager: Installing framework-zephyr-mipi-sys-t @ ~0.20200.191024\nframework-zephyr-mipi-sys-t @ 0.20200.191024 has been successfully installed!\nPackageManager: Installing framework-zephyr-canopennode @ ~0.20200.200115\nframework-zephyr-canopennode @ 0.20200.200115 has been successfully installed!\nPackageManager: Installing framework-zephyr-loramac-node @ ~0.20200.191220\nframework-zephyr-loramac-node @ 0.20200.191220 has been successfully installed!\nPackageManager: Installing framework-zephyr-mcuboot @ ~0.20200.200219\nframework-zephyr-mcuboot @ 0.20200.200219 has been successfully installed!\nPackageManager: Installing framework-zephyr-nrf-hw-models @ ~0.20200.0\nframework-zephyr-nrf-hw-models @ 0.20200.191115 has been successfully installed!\nPackageManager: Installing tool-cmake @ ~3.16.0\ntool-cmake @ 3.16.4 has been successfully installed!\nPackageManager: Installing tool-dtc @ ~1.4.7\ntool-dtc @ 1.4.7 has been successfully installed!\nPackageManager: Installing tool-ninja @ ^1.7.0\ntool-ninja @ 1.7.1 has been successfully installed!\nPackageManager: Installing tool-gperf @ ^3.0.0\ntool-gperf @ 3.0.4 has been successfully installed!\nCorePackageManager: Installing tool-scons @ ~3.30102.0\ntool-scons @ 3.30102.0 has been successfully installed!\n======================== [SUCCESS] Took 111.75 seconds ========================"

valeros, 2020-05-14:
I don't see any errors in the log. Have you tried to compile a simple project?

k0d3x8its, 2020-05-16:
Will check.