platformio/platformio-home #812: Could not install platform

zhu7284, 2020-05-20:
PIO Core Call Error: "PlatformManager: Installing \u001b[36mnordicnrf51\u001b[0m\nDownloading...\nUnpacking...\nnordicnrf51 @ 5.1.0 has been successfully installed!\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 325, in platform_install\n    force=force,\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 103, in install\n    force=force,\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 307, in install_packages\n    self.pm.install(name, version, silent=silent, force=force)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 686, in install\n    self.print_message(msg)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 608, in print_message\n    click.echo(\"%s: %s\" % (self.__class__.__name__, message), nl=nl)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo\n    \"%s\\n\" % (args[0] if args else \"\")\n  File \"c:\\users\\admin\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n    value.decode() if is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2020-05-21:
Please provide an output from the system terminal:
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```
To me it looks like is caused by Python 2.x. Could you please try to install Python 3.7
https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

valeros, 2020-05-22:
Fixed in the development version, please run `pio upgrade --dev` in the IDE terminal.

closed 2020-05-22 by valeros