platformio/platformio-home #816: Libraries: Could not update library

wahibmichael, 2020-05-21:
PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\wahib\\Documents\\PlatformIO\\Projects\\Smart Remote\\.pio\\libdeps\\esp12e\u001b[0m\nUpdating \u001b[36mPubSubClient\u001b[0m                    @ 2.7            \n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\n    lm.update(library, only_check=only_check)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 789, in update\n    click.echo(\"[%s]\" % (click.style(latest, fg=\"red\")))\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo\n    \"%s\\n\" % (args[0] if args else \"\")\n  File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n    value.decode() if is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2020-05-21:
Hi @wahibmichael ! Could you please switch to the latest development version of `platformio-core`? 
https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usedevelopmentpiocore

wahibmichael, 2020-05-21:
The update is done successfully through CLI
[image]

wahibmichael, 2020-05-21:
It always fails from library manager GUI

valeros, 2020-05-21:
Please provide an output from the CLI:
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

wahibmichael, 2020-05-21:
@valeros please find below 

echo %PATH%
%PATH%

echo %PYTHONPATH%
%PYTHONPATH%

python --version
Python 2.7.18

pip --version
pip 20.1 from c:\users\wahib\.platformio\penv\lib\site-packages\pip (python 2.7)

pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package          Version
---------------- ----------
bottle           0.12.18
certifi          2020.4.5.1
chardet          3.0.4
click            7.1.2
colorama         0.4.3
idna             2.9
marshmallow      2.21.0
pip              20.1
platformio       4.3.3
pyelftools       0.26
pyserial         3.4
requests         2.23.0
semantic-version 2.8.5
setuptools       44.1.0
tabulate         0.8.7
urllib3          1.25.9
wheel            0.34.2
WARNING: You are using pip version 20.1; however, version 20.1.1 is available.
You should consider upgrading via the 'c:\users\wahib\.platformio\penv\scripts\python.exe -m pip install --upgrade pip' command.

valeros, 2020-05-21:
Thanks! Looks like the issue only shows up on Python 2.x and it's already fixed in the development branch of `platformio-core`. Could you please try to enable it in the IDE:
File->Preferences->Settings->Extensions->PlatformIO IDE:
[image]

wahibmichael, 2020-05-21:
Setting updated, IDE restarted the same issue

PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\wahib\\Documents\\PlatformIO\\Projects\\Smart Remote\\.pio\\libdeps\\esp12e\u001b[0m\nUpdating \u001b[36mPubSubClient\u001b[0m @ 2.7 \n\n\nError: Traceback (most recent call last):\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main\n cli() # pylint: disable=no-value-for-parameter\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n return self.main(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\n lm.update(library, only_check=only_check)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\managers\\package.py\", line 789, in update\n click.echo(\"[%s]\" % (click.style(latest, fg=\"red\")))\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>\n click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo\n \"%s\\n\" % (args[0] if args else \"\")\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n value.decode() if is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n https://github.com/platformio/platformio-core/issues\n\n============================================================"

wahibmichael, 2020-05-21:
shall I upgrade the python version?

valeros, 2020-05-21:
Yes, the newer version will fix this, but I'm really interested in resolving this issue on Python 2.x. If you're willing to help we can try to hunt down this bug.
From the log you attached, it looks like you're still using the old `platformio-core`. What is the output for `platformio --version`

wahibmichael, 2020-05-21:
platformio --version
PlatformIO, version 4.3.3

wahibmichael, 2020-05-21:
[image]

valeros, 2020-05-21:
So it's still the same stable version. Could you please try to run in the terminal the following command:
```
pio upgrade --dev
```
The version should be something like this `version 4.3.4a4`

wahibmichael, 2020-05-21:
platformio --version
PlatformIO, version 4.3.4a5

same issue 

PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\wahib\\Documents\\PlatformIO\\Projects\\Smart Remote\\.pio\\libdeps\\esp12e\u001b[0m\nUpdating \u001b[36mPubSubClient\u001b[0m @ 2.7 \n\n\nError: Traceback (most recent call last):\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\n cli() # pylint: disable=no-value-for-parameter\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n return self.main(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\n lm.update(library, only_check=only_check)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\managers\\package.py\", line 793, in update\n click.echo(\"[%s]\" % (click.style(latest, fg=\"red\")))\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 97, in <lambda>\n click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 94, in _safe_echo\n \"%s\\n\" % (args[0] if args else \"\")\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n value.decode() if not PY2 and is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n https://github.com/platformio/platformio-core/issues\n\n============================================================"

wahibmichael, 2020-05-21:
updated to python 3.8
 python --version
Python 3.8.3

still same problem
PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\wahib\\Documents\\PlatformIO\\Projects\\Smart Remote\\.pio\\libdeps\\esp12e\u001b[0m\nUpdating \u001b[36mPubSubClient\u001b[0m @ 2.7 \n\n\nError: Traceback (most recent call last):\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\n cli() # pylint: disable=no-value-for-parameter\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n return self.main(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"C:\\Python27\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\n lm.update(library, only_check=only_check)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\managers\\package.py\", line 793, in update\n click.echo(\"[%s]\" % (click.style(latest, fg=\"red\")))\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 97, in <lambda>\n click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n File \"C:\\Python27\\lib\\site-packages\\platformio\\__main__.py\", line 94, in _safe_echo\n \"%s\\n\" % (args[0] if args else \"\")\n File \"C:\\Python27\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n value.decode() if not PY2 and is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n https://github.com/platformio/platformio-core/issues\n\n============================================================"

wahibmichael, 2020-05-22:
@valeros Any solution?

valeros, 2020-05-22:
Hi @wahibmichael ! Sorry for the late reply.

> updated to python 3.8
python --version
Python 3.8.3

> still same problem

I see from the log that you're still using python 2.7:
```
C:\Python27\lib\site-packages\platformio\managers\package.py
```
BTW, how did you install PlatformIO? It looks like it's installed globally in your system?

Anyway, could you please also try to pull the latest changes again via:
`pio upgrade --dev`

wahibmichael, 2020-05-22:
> Hi @wahibmichael ! Sorry for the late reply.
>
>
>
> I see from the log that you're still using python 2.7:
>
> ```
> C:\Python27\lib\site-packages\platformio\managers\package.py
> ```
>
> BTW, how did you install [...]

-Hello Installed PlatformIO from VScode extension manager

-upgraded the version 
platformio --version
PlatformIO, version 4.3.4a5

same issue 

PIO Core Call Error: "\u001b[1mLibrary Storage: \u001b[0m\u001b[34mC:\\Users\\wahib\\Documents\\PlatformIO\\Projects\\Smart Remote\\.pio\\libdeps\\esp12e\u001b[0m\nUpdating \u001b[36mPubSubClient\u001b[0m @ 2.7 \n\n\nError: Traceback (most recent call last):\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\n cli() # pylint: disable=no-value-for-parameter\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n return self.main(*args, **kwargs)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\n lm.update(library, only_check=only_check)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 793, in update\n click.echo(\"[%s]\" % (click.style(latest, fg=\"red\")))\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 97, in <lambda>\n click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 94, in _safe_echo\n \"%s\\n\" % (args[0] if args else \"\")\n File \"c:\\users\\wahib\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write\n value.decode() if not PY2 and is_bytes(value) else value\nTypeError: 'unicode' does not have the buffer interface\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n https://github.com/platformio/platformio-core/issues\n\n============================================================"

wahibmichael, 2020-05-22:
BTW I tried to install python 2.7 and re-installed Python 3.8, however, PlatformIO failed to setup

valeros, 2020-05-22:
@wahibmichael ! I still cannot confirm that you're using the latest version.
Could you please close the IDE, delete any global installations of PlatformIO, delete folder `c:\users\wahib\.platformio\penv`, open IDE, wait until a new installation is complete, reload window, open the terminal and pull the latest version via `pio upgrade --dev`

wahibmichael, 2020-05-22:
working now perfectly, thanks for your support 😄

closed 2020-05-22 by valeros