platformio/platformio-home #819: Could not import project

Ironman-007, 2020-05-22:
PIO Core Call Error: "The current working directory /home/ironman/Documents/PlatformIO/Projects/200521-214351-zephyr-blink will be used for the project.\n\nThe next files/directories have been created in /home/ironman/Documents/PlatformIO/Projects/200521-214351-zephyr-blink\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing adafruit_feather_nrf52832 (platform: nordicnrf52; framework: zephyr; board: adafruit_feather_nrf52832)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html\nPLATFORM: Nordic nRF52 4.2.0 > Adafruit Bluefruit nRF52832 Feather\nHARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash\nDEBUG: Current (jlink) External (jlink, stlink)\nPACKAGES: \n - framework-zephyr 2.20200.200401 (2.2.0) \n - framework-zephyr-canopennode 0.20200.200115 (2.2.0) \n - framework-zephyr-civetweb 0.20200.190807 (2.2.0) \n - framework-zephyr-fatfs 0.20200.191125 (2.2.0) \n - framework-zephyr-hal-nordic 0.20200.200130 (2.2.0) \n - framework-zephyr-libmetal 0.20200.190530 (2.2.0) \n - framework-zephyr-littlefs 0.20200.190811 (2.2.0) \n - framework-zephyr-loramac-node 0.20200.191220 (2.2.0) \n - framework-zephyr-lvgl 0.20200.191207 (2.2.0) \n - framework-zephyr-mbedtls 0.20200.200206 (2.2.0) \n - framework-zephyr-mcuboot 0.20200.200219 (2.2.0) \n - framework-zephyr-mcumgr 0.20200.200117 (2.2.0) \n - framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0) \n - framework-zephyr-nrf-hw-models 0.20200.191115 (2.2.0) \n - framework-zephyr-open-amp 0.20200.190612 (2.2.0) \n - framework-zephyr-openthread 0.20200.191014 (2.2.0) \n - framework-zephyr-segger 0.20200.190421 (2.2.0) \n - framework-zephyr-tinycbor 0.20200.191016 (2.2.0) \n - tool-cmake 3.16.4 \n - tool-dtc 1.4.7 \n - tool-gperf 3.0.4 \n - tool-ninja 1.7.1 \n - tool-sreccat 1.164.0 (1.64) \n - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)\nPython 3.4 or greater required, but you have Python 2.7.18\n========================== [FAILED] Took 0.60 seconds ==========================\n\nEnvironment                Status    Duration\n-------------------------  --------  ------------\nadafruit_feather_nrf52832  FAILED    00:00:00.600\nnrf52_dk                   IGNORED\nnrf52840_dk                IGNORED\nparticle_xenon             IGNORED\n==================== 1 failed, 0 succeeded in 00:00:00.600 ===================="

Ironman-007, 2020-05-22:
I already changed the python interpreter to Python 3.5

valeros, 2020-06-25:
Looks like your PlatformIO virtual environment is still using Python 2.7, the Zephyr framework only supports Python 3.
- Close the IDE
- Make sure Python 3.x is the main Python installation in your system.
- Delete folder `/home/ironman/.platformio/.penv`
- Open the IDE and wait until a new installation is completed

closed 2020-06-25 by valeros