platformio/platformio-home #831: Could not load boards list

Merdock1, 2020-05-25:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 31, in cli\r\n    return _print_boards_json(query, installed)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 80, in _print_boards_json\r\n    for board in _get_boards(installed):\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 75, in _get_boards\r\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 205, in get_all_boards\r\n    boards = self.get_installed_boards()\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 59, in wrapper\r\n    self.cache[key] = (time.time(), func(*args, **kwargs))\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 193, in get_installed_boards\r\n    p = PlatformFactory.newPlatform(manifest[\"__pkg_dir\"])\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 273, in newPlatform\r\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 562, in __init__\r\n    self._manifest = fs.load_json(manifest_path)\r\n  File \"c:\\users\\nego\\.platformio\\penv\\lib\\site-packages\\platformio\\fs.py\", line 53, in load_json\r\n    with open(file_path, \"r\") as f:\r\nFileNotFoundError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\nego\\\\.platformio\\\\platforms\\\\Arduino_Core_STM32\\\\platform.json'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

valeros, 2020-05-25:
Hi @Merdock1, please delete folder `C:\\Users\\nego\\.platformio\\platforms\\Arduino_Core_STM32`.
`C:\\Users\\nego\\.platformio\\platforms` is reserved folder and should only contain properly installed platforms.

closed 2020-05-25 by valeros

Merdock1, 2020-05-25:
It was already done.
The report was sent by accident,
Thank you.