platformio/platformio-home #847: Could not initialize project

triss-tucker, 2020-05-29:
PIO Core Call Error: "The current working directory E:\\dev\\embedded\\zephyr\\z769_test will be used for the project.\n\r\nThe next files/directories have been created in E:\\dev\\embedded\\zephyr\\z769_test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing disco_f769ni (platform: ststm32; board: disco_f769ni; framework: zephyr)\n-------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f769ni.html\nPLATFORM: ST STM32 6.1.0 > ST 32F769IDISCOVERY\nHARDWARE: STM32F769NIH6 216MHz, 512KB RAM, 1MB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)\nPACKAGES: \n - framework-zephyr 2.20200.200401 (2.2.0) \n - framework-zephyr-canopennode 0.20200.200115 (2.2.0) \n - framework-zephyr-civetweb 0.20200.190807 (2.2.0) \n - framework-zephyr-fatfs 0.20200.191125 (2.2.0) \n - framework-zephyr-hal-st 0.20200.190924 (2.2.0) \n - framework-zephyr-hal-stm32 0.20200.200206 (2.2.0) \n - framework-zephyr-libmetal 0.20200.190530 (2.2.0) \n - framework-zephyr-littlefs 0.20200.190811 (2.2.0) \n - framework-zephyr-loramac-node 0.20200.191220 (2.2.0) \n - framework-zephyr-lvgl 0.20200.191207 (2.2.0) \n - framework-zephyr-mbedtls 0.20200.200206 (2.2.0) \n - framework-zephyr-mcuboot 0.20200.200219 (2.2.0) \n - framework-zephyr-mcumgr 0.20200.200117 (2.2.0) \n - framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0) \n - framework-zephyr-open-amp 0.20200.190612 (2.2.0) \n - framework-zephyr-openthread 0.20200.191014 (2.2.0) \n - framework-zephyr-segger 0.20200.190421 (2.2.0) \n - framework-zephyr-tinycbor 0.20200.191016 (2.2.0) \n - tool-cmake 3.16.4 \n - tool-dtc 1.4.7 \n - tool-ninja 1.9.0 \n - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)\nPython 3.4 or greater required, but you have Python 2.7.15\n========================= [FAILED] Took 3.82 seconds ========================="

When I open terminal and issue 'python --version' it returns 'Python 3.8.0'

path is 
PATH=C:\Program Files\AdoptOpenJDK\jdk-8.0.252.09-hotspot\bin;C:\Program Files\AdoptOpenJDK\jre-8.0.252.09-hotspot\bin;C:\Program Files\AdoptOpenJDK\jdk-11.0.4.11-hotspot\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\OpenOCD\bin;C:\Program Files\dotnet\;C:\Program Files\PuTTY\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Git\cmd;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\Calibre2\;C:\Go\bin;C:\Program Files\nodejs\;C:\Program Files (x86)\Yarn\bin\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\Mercurial;c:\dev\bin;C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin;C:\Users\Triss\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\Triss\AppData\Local\Programs\Python\Python38\;C:\Users\Triss\AppData\Local\Microsoft\WindowsApps;C:\Users\Triss\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Triss\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\Users\Triss\go\bin;C:\Users\Triss\AppData\Roaming\npm;C:\Users\Triss\AppData\Local\Yarn\bin;

valeros, 2020-05-29:
Looks like your PlatformIO virtual environment was created for Python 2.7.
- Close the IDE
- Make sure Python 3.x is the main Python installation in your system.
- Delete folder `C:\Users\Triss\.platformio\.penv`
- Open IDE and wait until a new installation is completed

triss-tucker, 2020-05-29:
That worked fine, thanks Valeros

closed 2020-05-29 by triss-tucker