platformio/platformio-home #870: Could not initialize project

larsfriday, 2020-06-02:
PIO Core Call Error: "The current working directory C:\\Users\\larsfleckenstein\\Documents\\PlatformIO\\Projects\\Friday_Zephyr_FW_LOCK will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\larsfleckenstein\\Documents\\PlatformIO\\Projects\\Friday_Zephyr_FW_LOCK\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing nrf52840_dk (platform: nordicnrf52; board: nrf52840_dk; framework: zephyr)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html\r\nPLATFORM: Nordic nRF52 4.2.0 > Nordic nRF52840-DK\r\nHARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash\r\nDEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)\r\nPACKAGES: \r\n - framework-zephyr 2.20200.200401 (2.2.0) \r\n - framework-zephyr-canopennode 0.20200.200115 (2.2.0) \r\n - framework-zephyr-civetweb 0.20200.190807 (2.2.0) \r\n - framework-zephyr-fatfs 0.20200.191125 (2.2.0) \r\n - framework-zephyr-hal-nordic 0.20200.200130 (2.2.0) \r\n - framework-zephyr-libmetal 0.20200.190530 (2.2.0) \r\n - framework-zephyr-littlefs 0.20200.190811 (2.2.0) \r\n - framework-zephyr-loramac-node 0.20200.191220 (2.2.0) \r\n - framework-zephyr-lvgl 0.20200.191207 (2.2.0) \r\n - framework-zephyr-mbedtls 0.20200.200206 (2.2.0) \r\n - framework-zephyr-mcuboot 0.20200.200219 (2.2.0) \r\n - framework-zephyr-mcumgr 0.20200.200117 (2.2.0) \r\n - framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0) \r\n - framework-zephyr-nrf-hw-models 0.20200.191115 (2.2.0) \r\n - framework-zephyr-open-amp 0.20200.190612 (2.2.0) \r\n - framework-zephyr-openthread 0.20200.191014 (2.2.0) \r\n - framework-zephyr-segger 0.20200.190421 (2.2.0) \r\n - framework-zephyr-tinycbor 0.20200.191016 (2.2.0) \r\n - tool-cmake 3.16.4 \r\n - tool-dtc 1.4.7 \r\n - tool-ninja 1.9.0 \r\n - tool-sreccat 1.164.0 (1.64) \r\n - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)\r\nReading CMake configuration...\r\n-- Zephyr version: 2.2.0\r\n-- Found PythonInterp: c:/users/larsfleckenstein/.platformio/penv/scripts/python.exe (found suitable version \"3.8.2\", minimum required is \"3.6\") \r\n-- Selected BOARD nrf52840_pca10056\r\n-- Loading C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base\r\n-- Configuring incomplete, errors occurred!\r\n\r\nCMake Error at C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/generic_toolchain.cmake:64 (include):\r\n  include could not find load file:\r\n\r\n    C:\\dev\\vGDB\\ToolChain\\sysGCC\\arm-eabi/cmake/toolchain/gnuarmemb/generic.cmake\r\nCall Stack (most recent call first):\r\n  C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:462 (include)\r\n  CMakeLists.txt:2 (include)\r\n\r\n\r\nCMake Error at C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/dts.cmake:140 (message):\r\n  command failed with return code: The system cannot find the file specified\r\nCall Stack (most recent call first):\r\n  C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:463 (include)\r\n  CMakeLists.txt:2 (include)\r\n\r\n\r\n========================== [FAILED] Took 4.35 seconds =========================="

larsfriday, 2020-06-02:
trying to make my first project after installing Platformio in VScode

valeros, 2020-06-02:
Hi @larsfriday ! Do you have any global Zephyr installations in your system? Please run `pio run -t envdump` in the IDE terminal and attach the output here.

larsfriday, 2020-06-02:
Hi Valerii

Thank for your prompt reply.

It look like I messed up the whole platformio system from the begining ☹
I have installed and uninstall the platformio extension several times now.
Now I cannot run platformio or pio from the terminal - Cannot find it. This was possible before the uninstall and install.

HOW CAN I GET BACK to a clean stage.

I use VisualGDB and want to move to platformio.
My ToolChain env vars are wrong. Can that be the reason for my reported issue?

This is my envs
!::=::\
!C:=C:\Users\larsfleckenstein
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\larsfleckenstein\AppData\Roaming
BOOST_ROOT=C:\dev\boost_1_72_0\
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=132323717342629683
COMMONPROGRAMFILES=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=DESKTOP-VJL3ROH
COMSPEC=C:\WINDOWS\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
HOMEDRIVE=C:
HOMEPATH=\Users\larsfleckenstein
LOCALAPPDATA=C:\Users\larsfleckenstein\AppData\Local
LOGONSERVER=\\DESKTOP-VJL3ROH
NUMBER_OF_PROCESSORS=8
OneDrive=C:\Users\larsfleckenstein\OneDrive - Friday Home
OneDriveCommercial=C:\Users\larsfleckenstein\OneDrive - Friday Home
OneDriveConsumer=C:\Users\larsfleckenstein\Documents\OneLarsFleckPrivat\OneDrive
OS=Windows_NT
PATH=/cygdrive/c/Python38/Scripts:/cygdrive/c/Python38:/cygdrive/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Microsoft Visual Studio/Shared/Python37_64/Scripts:/cygdrive/c/Program Files (x86)/Microsoft Visual Studio/Shared/Python37_64:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files/dotnet:/cygdrive/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn:/cygdrive/c/Program Files/otii:/cygdrive/c/Program Files (x86)/Microsoft SQL Server/150/DTS/Binn:/cygdrive/c/Python27:/cygdrive/c/Program Files/Microsoft SQL Server/110/Tools/Binn:/cygdrive/c/Program Files/PuTTY:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/ProgramData/chocolatey/bin:/cygdrive/c/Program Files/CMake/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/Docker/Docker/resources/bin:/cygdrive/c/ProgramData/DockerDesktop/version-bin:/cygdrive/c/Program Files (x86)/Nordic Semiconductor/nrf-command-line-tools/bin:/cygdrive/c/Ruby26-x64/bin:/cygdrive/c/Users/larsfleckenstein/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/larsfleckenstein/AppData/Local/Programs/Microsoft VS Code/bin:/cygdrive/c/Users/larsfleckenstein/AppData/Local/Programs/Azure Data Studio/bin:/cygdrive/c/Users/larsfleckenstein/.dotnet/tools:/usr/bin:/cygdrive/c/mingw/bin:/cygdrive/c/Users/larsfleckenstein/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/dev/vGDB/ToolChain/sysGCC/arm-eabi/lib/gcc/arm-none-eabi/9.2.1:/cygdrive/c/Program Files/Multipass/bin
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW;.RB;.RBW
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 142 Stepping 10, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=8e0a
ProgramData=C:\ProgramData
PROGRAMFILES=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SYSTEMDRIVE=C:
SYSTEMROOT=C:\WINDOWS
TEMP=/cygdrive/c/Users/LARSFL~1/AppData/Local/Temp
TMP=/cygdrive/c/Users/LARSFL~1/AppData/Local/Temp
ToolChainDir=C:\dev\vGDB\ToolChain\sysGCC\arm-eabi
ToolchainProfile2=C:\dev\vGDB\ToolChain\sysGCC\arm-eabi
TOOLCHAIN_ROOT=C:\dev\vGDB\ToolChain\sysGCC\arm-eabi
USERDOMAIN=AzureAD
USERDOMAIN_ROAMINGPROFILE=AzureAD
USERNAME=larsfleckenstein
USERPROFILE=C:\Users\larsfleckenstein
VISUALGDB_DIR=C:\Program Files (x86)\Sysprogs\VisualGDB\
WINDIR=C:\WINDOWS
TERM=cygwin
HOME=/home/larsfleckenstein

lars

larsfriday, 2020-06-02:
I did this and the pio cmd runs .-)
Windows<https://docs.platformio.org/en/latest/core/installation.html#id42>
Please read one of these instructions How do I set or change the PATH system variable?<https://www.google.com.ua/search?q=how+do+i+set+or+change+the+path+system+variable>
You need to edit system environment variable called Path and append C:\Users\UserName\.platformio\penv\Scripts; path in the beginning of a list (please replace UserName with your account name).
outPUT

PS C:\Users\larsfleckenstein\Documents\PlatformIO\Projects\200513-083910-hello-world> pio run -t envdump
Processing windows_x86 (platform: windows_x86)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
PACKAGES:
 - toolchain-gccmingw32 1.50100.0 (5.1.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
{ 'AR': '${_BINPREFIX}ar',
  'ARCOM': '$AR $ARFLAGS $TARGET ${_long_sources_hook(__env__, SOURCES)}',
  'ARCOMSTR': 'Archiving $TARGET',
  'ARFLAGS': ['rc'],
  'AS': '$CC',
  'ASCOM': '$ASPPCOM',
  'ASCOMSTR': 'Compiling $TARGET',
  'ASFLAGS': [],
  'ASPPCOM': '$CC $ASPPFLAGS $CPPFLAGS $_CPPDEFFLAGS '
             '${_long_incflags_hook(__env__, _CPPINCFLAGS)} -c -o $TARGET '
             '$SOURCES',
  'ASPPCOMSTR': 'Compiling $TARGET',
  'ASPPFLAGS': '$ASFLAGS',
  'BUILDERS': { 'Library': <SCons.Builder.BuilderBase object at 0x00000193A1BCA1C0>,
                'LoadableModule': <SCons.Builder.BuilderBase object at 0x00000193A1BD8DF0>,
                'Object': <SCons.Builder.CompositeBuilder object at 0x00000193A1BCA490>,
                'Program': <SCons.Builder.BuilderBase object at 0x00000193A1BD8790>,
                'SharedLibrary': <SCons.Builder.BuilderBase object at 0x00000193A1BD82B0>,
                'SharedObject': <SCons.Builder.CompositeBuilder object at 0x00000193A1BCA6A0>,
                'StaticLibrary': <SCons.Builder.BuilderBase object at 0x00000193A1BCA1C0>,
                'StaticObject': <SCons.Builder.CompositeBuilder object at 0x00000193A1BCA490>},
  'BUILD_CACHE_DIR': None,
  'BUILD_DIR': '$PROJECT_BUILD_DIR\\$PIOENV',
  'BUILD_SCRIPT': 'C:\\Users\\larsfleckenstein\\.platformio\\platforms\\windows_x86\\builder\\main.py',
  'BUILD_SRC_DIR': '$BUILD_DIR\\src',
  'BUILD_TEST_DIR': '$BUILD_DIR\\test',
  'CC': '${_BINPREFIX}gcc',
  'CCCOM': '$CC -o $TARGET -c $CFLAGS $CCFLAGS $_CCCOMCOM $SOURCES',
  'CCCOMSTR': 'Compiling $TARGET',
  'CCFLAGS': [],
  'CFILESUFFIX': '.c',
  'CFLAGS': [],
  'COMPILATIONDB_PATH': '$BUILD_DIR\\compile_commands.json',
  'CONFIGUREDIR': '#/.sconf_temp',
  'CONFIGURELOG': '#/config.log',
  'CPPDEFINES': [('PLATFORMIO', 40304)],
  'CPPDEFPREFIX': '-D',
  'CPPDEFSUFFIX': '',
  'CPPSUFFIXES': [ '.c',
                   '.C',
                   '.cxx',
                   '.cpp',
                   '.c++',
                   '.cc',
                   '.h',
                   '.H',
                   '.hxx',
                   '.hpp',
                   '.hh',
                   '.F',
                   '.fpp',
[... 127 lines cut ...]
             'piomisc',
             'pioide',
             'piosize'],
  'UNIX_TIME': 1591133261,
  '_BINPREFIX': '',
  '_CCCOMCOM': '$CPPFLAGS $_CPPDEFFLAGS ${_long_incflags_hook(__env__, '
               '_CPPINCFLAGS)}',
  '_CPPDEFFLAGS': '${_defines(CPPDEFPREFIX, CPPDEFINES, CPPDEFSUFFIX, '
                  '__env__)}',
  '_CPPINCFLAGS': '$( ${_concat(INCPREFIX, CPPPATH, INCSUFFIX, __env__, RDirs, '
                  'TARGET, SOURCE)} $)',
  '_LIBDIRFLAGS': '$( ${_concat(LIBDIRPREFIX, LIBPATH, LIBDIRSUFFIX, __env__, '
                  'RDirs, TARGET, SOURCE)} $)',
  '_LIBFLAGS': '${_stripixes(LIBLINKPREFIX, LIBS, LIBLINKSUFFIX, LIBPREFIXES, '
               'LIBSUFFIXES, __env__)}',
  '__DSHLIBVERSIONFLAGS': '${__libversionflags(__env__,"DSHLIBVERSION","_DSHLIBVERSIONFLAGS")}',
  '__LDMODULEVERSIONFLAGS': '${__libversionflags(__env__,"LDMODULEVERSION","_LDMODULEVERSIONFLAGS")}',
  '__PIO_LIB_BUILDERS': [],
  '__SHLIBVERSIONFLAGS': '${__libversionflags(__env__,"SHLIBVERSION","_SHLIBVERSIONFLAGS")}',
  '__libversionflags': <function __libversionflags at 0x00000193A1560940>,
  '_concat': <function _concat at 0x00000193A1560670>,
  '_defines': <function _defines at 0x00000193A15608B0>,
  '_long_incflags_hook': <function long_incflags_hook at 0x00000193A1BEFEE0>,
  '_long_sources_hook': <function long_sources_hook at 0x00000193A1BEFE50>,
  '_stripixes': <function _stripixes at 0x00000193A1560790>,

larsfriday, 2020-06-02:
Running the nrf example zephyr-ble-beacon

Fails with
  ModuleNotFoundError: No module named 'six'

C:\dev\sandbox\platform-nordicnrf52-develop\platform-nordicnrf52-develop\examples\zephyr-ble-beacon>pio run
Processing nrf52840_dk (platform: nordicnrf52; framework: zephyr; board: nrf52840_dk)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 4.2.0 > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES:
- framework-zephyr 2.20200.200401 (2.2.0)
- framework-zephyr-canopennode 0.20200.200115 (2.2.0)
- framework-zephyr-civetweb 0.20200.190807 (2.2.0)
- framework-zephyr-fatfs 0.20200.191125 (2.2.0)
- framework-zephyr-hal-nordic 0.20200.200130 (2.2.0)
- framework-zephyr-libmetal 0.20200.190530 (2.2.0)
- framework-zephyr-littlefs 0.20200.190811 (2.2.0)
- framework-zephyr-loramac-node 0.20200.191220 (2.2.0)
- framework-zephyr-lvgl 0.20200.191207 (2.2.0)
- framework-zephyr-mbedtls 0.20200.200206 (2.2.0)
- framework-zephyr-mcuboot 0.20200.200219 (2.2.0)
- framework-zephyr-mcumgr 0.20200.200117 (2.2.0)
- framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0)
- framework-zephyr-nrf-hw-models 0.20200.191115 (2.2.0)
- framework-zephyr-open-amp 0.20200.190612 (2.2.0)
- framework-zephyr-openthread 0.20200.191014 (2.2.0)
- framework-zephyr-segger 0.20200.190421 (2.2.0)
- framework-zephyr-tinycbor 0.20200.191016 (2.2.0)
- tool-cmake 3.16.4
- tool-dtc 1.4.7
- tool-ninja 1.9.0
- tool-sreccat 1.164.0 (1.64)
- toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
Reading CMake configuration...
-- Zephyr version: 2.2.0
-- Found PythonInterp: c:/users/larsfleckenstein/.platformio/penv/scripts/python.exe (found suitable version "3.8.2", minimum required is "3.6")
-- Selected BOARD nrf52840_pca10056
-- Configuring incomplete, errors occurred!

CMake Error at C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/zephyr_module.cmake:45 (message):
  Traceback (most recent call last):

    File "C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/scripts/zephyr_module.py", line 24, in <module>
      import pykwalify.core
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\pykwalify\core.py", line 25, in <module>
      from dateutil.parser import parse
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\__init__.py", line 2, in <module>
      from ._parser import parse, parser, parserinfo, ParserError
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\_parser.py", line 42, in <module>
      import six

  ModuleNotFoundError: No module named 'six'

Call Stack (most recent call first):
  C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:461 (include)
  CMakeLists.txt:4 (include)

============================================== [FAILED] Took 3.23 seconds ==============================================

Processing particle_xenon (platform: nordicnrf52; framework: zephyr; board: particle_xenon)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/particle_xenon.html
PLATFORM: Nordic nRF52 4.2.0 > Particle Xenon
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
[... 35 lines cut ...]
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\pykwalify\core.py", line 25, in <module>
      from dateutil.parser import parse
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\__init__.py", line 2, in <module>
      from ._parser import parse, parser, parserinfo, ParserError
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\_parser.py", line 42, in <module>
      import six

  ModuleNotFoundError: No module named 'six'

Call Stack (most recent call first):
  C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:461 (include)
  CMakeLists.txt:4 (include)

============================================== [FAILED] Took 3.07 seconds ==============================================

Environment     Status    Duration
--------------  --------  ------------
nrf52840_dk     FAILED    00:00:03.231
particle_xenon  FAILED    00:00:03.071
======================================== 2 failed, 0 succeeded in 00:00:06.303 ========================================

C:\dev\sandbox\platform-nordicnrf52-develop\platform-nordicnrf52-develop\examples\zephyr-ble-beacon>

  'ENV': environ({'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\larsfleckenstein\\AppData\\Roaming', 'BOOST_ROOT': 'C:\\dev\\boost_1_72_0\\', 'CHOCOLATEYINSTALL': 'C:\\ProgramData\\chocolatey', 'CHOCOLATEYLASTPATHUPDATE': '132323717342629683', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common<file://Common> Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'DESKTOP-VJL3ROH',
[... cut ...] M': 'truecolor', 'VSCODE_GIT_IPC_HANDLE': '\\\\.\\pipe\\vscode-git-fe1733686d-sock', 'GIT_ASKPASS': 'c:\\Users\\larsfleckenstein\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\extensions\\git\\dist\\askpass.sh', 'VSCODE_GIT_ASKPASS_NODE': 'C:\\Users\\larsfleckenstein\\AppData\\Local\\Programs\\Microsoft VS Code\\Code.exe', 'VSCODE_GIT_ASKPASS_MAIN': 'c:\\Users\\larsfleckenstein\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\extensions\\git\\dist\\askpass-main.js'}),

valeros, 2020-06-02:
Try to run the following command in the IDE terminal:
```
c:\\users\\larsfleckenstein\\.platformio\\penv\\scripts\\pip.exe install six
```
Then clean your project and try to compile again.
Also I'm not sure about  `TOOLCHAIN_ROOT` variable. You can try to unset it:
```
set TOOLCHAIN_ROOT=

```

larsfriday, 2020-06-02:
Hi Valerii

Thank you for your guidance.

I cannot make sense of this error messages. It state that it cannot find the module six which is installed. ☹. So it must be something different.

Lars

Dump from terminal.

The module six is installed
PS C:\dev\sandbox\platform-nordicnrf52-develop\platform-nordicnrf52-develop\examples\zephyr-ble-beacon> pip install six
Requirement already satisfied: six in c:\python38\lib\site-packages (1.15.0)

PS C:\dev\sandbox\platform-nordicnrf52-develop\platform-nordicnrf52-develop\examples\zephyr-ble-beacon> pio run -t envdump
Processing nrf52840_dk (platform: nordicnrf52; framework: zephyr; board: nrf52840_dk)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk.html
PLATFORM: Nordic nRF52 4.2.0 > Nordic nRF52840-DK
HARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)
PACKAGES:
- framework-zephyr 2.20200.200401 (2.2.0)
- framework-zephyr-canopennode 0.20200.200115 (2.2.0)
- framework-zephyr-civetweb 0.20200.190807 (2.2.0)
 - framework-zephyr-fatfs 0.20200.191125 (2.2.0)
- framework-zephyr-hal-nordic 0.20200.200130 (2.2.0)
- framework-zephyr-libmetal 0.20200.190530 (2.2.0)
- framework-zephyr-littlefs 0.20200.190811 (2.2.0)
 - framework-zephyr-loramac-node 0.20200.191220 (2.2.0)
- framework-zephyr-lvgl 0.20200.191207 (2.2.0)
- framework-zephyr-mbedtls 0.20200.200206 (2.2.0)
- framework-zephyr-mcuboot 0.20200.200219 (2.2.0)
- framework-zephyr-mcumgr 0.20200.200117 (2.2.0)
 - framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0)
- framework-zephyr-nrf-hw-models 0.20200.191115 (2.2.0)
- framework-zephyr-open-amp 0.20200.190612 (2.2.0)
- framework-zephyr-openthread 0.20200.191014 (2.2.0)
- framework-zephyr-segger 0.20200.190421 (2.2.0)
- framework-zephyr-tinycbor 0.20200.191016 (2.2.0)
 - tool-cmake 3.16.4
- tool-dtc 1.4.7
- tool-ninja 1.9.0
- tool-sreccat 1.164.0 (1.64)
- toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
Reading CMake configuration...
-- Zephyr version: 2.2.0
-- Selected BOARD nrf52840_pca10056
-- Configuring incomplete, errors occurred!

CMake Error at C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/zephyr_module.cmake:45 (message):
  Traceback (most recent call last):

    File "C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/scripts/zephyr_module.py", line 24, in <module>
      import pykwalify.core
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\pykwalify\core.py", line 25, in <module>
      from dateutil.parser import parse
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\__init__.py", line 2, in <module>
      from ._parser import parse, parser, parserinfo, ParserError
    File "c:\users\larsfleckenstein\.platformio\penv\lib\site-packages\dateutil\parser\_parser.py", line 42, in <module>
      import six

  ModuleNotFoundError: No module named 'six'

Call Stack (most recent call first):
  C:/Users/larsfleckenstein/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:461 (include)
  CMakeLists.txt:4 (include)

valeros, 2020-06-03:
The command for six was a bit different, you need to install this package in PlatformIO virtual environment:
```
c:\\users\\larsfleckenstein\\.platformio\\penv\\scripts\\pip.exe install six
```

closed 2020-09-16 by stale[bot]