platformio/platformio-home #889: Could not initialize project

iftoz, 2020-06-07:
Hello,
Im new to this VS code and PIO when i fisrt time creating new project in PIO it got stuck for a long time in please wait loading and finaly got this eror message :

PIO Core Call Error: "The current working directory C:\\Users\\Mifth\\Documents\\PlatformIO\\Projects\\newworld8266 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Mifth\\Documents\\PlatformIO\\Projects\\newworld8266\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError:"

i try to downgrame PIO version and reinstally few times but still got the same stuck issue, is there any way to fix it? im using windows 7 by the way

valeros, 2020-06-07:
Hi @iftoz ! It's hard to guess since there is no actual error message in your report. Try to create a project from PlatformO Home page.

iftoz, 2020-06-07:
Hi @valeros , i did create project from PIO homepage
here's what i got from toggle developer tools console for another details : 
 
[image]

im using windows x64 here but why it says "windows_x86" in the eror message?

valeros, 2020-06-08:
That's strange, there shouldn't be platform requirements for this package. Could you please try to import a project using `Project Examples` button. And then try to run `pio run` in the IDE terminal.

alextrof94, 2020-06-11:
Hello. Same problem.
Tried create project. Tried create project in nondefault directory.

Also tried import:
```
Could not import project
PIO Core Call Error: "The current working directory C:\\Users\\alext\\OneDrive\\Документы\\PlatformIO\\Projects\\200611-132738-arduino-blink will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\alext\\OneDrive\\Документы\\PlatformIO\\Projects\\200611-132738-arduino-blink\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing lightblue-bean (platform: atmelavr; framework: arduino; board: lightblue-bean)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\n File \"C:\\Users\\alext\\.platformio\\platforms\\atmelavr\\builder\\main.py\", line 30\r\n\r\n print \"Please unplug/plug device ...\"\r\n\r\n ^\r\n\r\nSyntaxError: Missing parentheses in call to 'print'. Did you mean print(\"Please unplug/plug device ...\")?\r\n\r\n========================== [FAILED] Took 1.09 seconds ==========================\r\n\r\nEnvironment Status Duration\r\n-------------------- -------- ------------\r\nlightblue-bean FAILED 00:00:01.089\r\nbluefruitmicro IGNORED\r\ndigispark-pro IGNORED\r\ndigispark-pro32 IGNORED\r\ndigispark-pro64 IGNORED\r\npro8MHzatmega328 IGNORED\r\nengduinov3 IGNORED\r\nethernet IGNORED\r\nfeather32u4 IGNORED\r\nflora8 IGNORED\r\nleonardo IGNORED\r\nmegaatmega2560 IGNORED\r\nmicro IGNORED\r\nmightycore1284 IGNORED\r\nmoteinomega IGNORED\r\npanStampAVR IGNORED\r\n1284p16m IGNORED\r\nquirkbot IGNORED\r\nreprap_rambo IGNORED\r\nrobotControl IGNORED\r\nrobotMotor IGNORED\r\nsanguino_atmega1284p IGNORED\r\nsparkfun_promicro16 IGNORED\r\nuno IGNORED\r\ngemma IGNORED\r\nattiny24 IGNORED\r\nwildfirev3 IGNORED\r\nyun IGNORED\r\natmega328pb IGNORED\r\n==================== 1 failed, 0 succeeded in 00:00:01.089 ===================="

```

valeros, 2020-06-11:
Thanks, the issue is caused by an outdated development platform on Python 3.

closed 2020-06-11 by valeros

LucasCanete, 2021-05-07:
@valeros could you further explain the solution? What must be done? Update python to its newest Version?

valeros, 2021-05-07:
@LucasCanete If you see the same error, then you need to install the latest Python.

LucasCanete, 2021-05-07:
Will try it out. Thanks

chomchom69420, 2023-07-31:
@iftoz Hi, sorry for bringing this up after such a long time. I am facing the same error of the project taking too long to build. I have also opened an [issue](https://github.com/platformio/platformio-core/issues/4705) regarding this. 
How did you get the error message on the console? I have opened the developer tools but haven't been able to get the error message yet. Is it displayed only when the build process crashes after a certain timeout? I am attaching a screenshot of what I can see.

[image]

AhvianRF, 2023-10-03:
Hi, have you found a solution to your problem?

I'm facing the same issue. I tried using platformio with vs Code for my projects two or three years ago, and each time I start a new project, it took ages at the "Project Wizard" thing.

Until now, I'm still facing the same issue with a different computer. I though it was a permission issue at first, but running vs code with admin didn't change a thing.

Now, I use a portable version of vs code and set up /.platformio at the portable directory (under the data folder), so I should get rid of permission problems, but still nothing.

Anytime I start a new project, it downloads the firmware at %\.platformio\platforms, creates the file architecture but there are only readme files in each folders, without main.cpp and platformio.ini

Tried any workarounds but no results. This is really frustrating.

[image]