platformio/platformio-home #893: Could not load platform data(Installed Atmel SAM)

phillowcompiler, 2020-06-08:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/platform.py\", line 235, in platform_show\n    data = _get_platform_data(platform)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/platform.py\", line 67, in _get_platform_data\n    return _get_installed_platform_data(*args, **kwargs)\n  File \"/home/matu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/platform.py\", line 101, in _get_installed_platform_data\n    data[\"boards\"] = [c.get_brief_data() for c in p.get_boards().values()]\n  File \"/home/matu/.platformio/platforms/atmelsam/platform.py\", line 105, in get_boards\n    result[key] = self._add_default_debug_tools(result[key])\n  File \"/home/matu/.platformio/platforms/atmelsam/platform.py\", line 148, in _add_default_debug_tools\n    assert openocd_chipname\nAssertionError\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2020-06-08:
Hi @phillowcompiler ! When exactly did this error happen? Just after installation?

phillowcompiler, 2020-06-08:
yes, after installation "Atmel SAM" platforms, and Clicked "Atmel SAM" at Platforms.
[image]

phillowcompiler, 2020-06-09:
Hi @valeros,
I tried this again, removed .platformio/penv and .platformio/platforms/atmelsam.
...it happened in Linux Mint( python --version = 3.6.9 ).
But There was no problem on Windows...

thanks

phillowcompiler, 2020-06-13:
Once I have removed everything under ~/.platfomio, it worked.

closed 2020-06-13 by phillowcompiler