platformio/platformio-home #903: Could not initialize project

iachievedit, 2020-06-14:
PIO Core Call Error: "The current working directory /Users/x/Documents/PlatformIO/Projects/freertos_i2c will be used for the project.\n\nThe next files/directories have been created in /Users/x/Documents/PlatformIO/Projects/freertos_i2c\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nPlatformManager: Installing sifive\nsifive @ 2.2.0 has been successfully installed!\nPackageManager: Installing toolchain-riscv @ ~1.80300.0\ntoolchain-riscv @ 1.80300.190927 has been successfully installed!\nThe platform 'sifive' has been successfully installed!\nThe rest of packages will be installed automatically depending on your build environment.\n\n\nError: Processing hifive1-revb (platform: sifive; board: hifive1-revb; framework: freertos)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nError: FreeRTOS cannot be used as a standalone framework without SiFive Freedom E SDK.\n========================== [FAILED] Took 0.90 seconds =========================="

valeros, 2020-06-14:
Hi @iachievedit ! At the moment, FreeRTOS can only be used in pair with Freedom E SDK https://docs.platformio.org/en/latest/frameworks/freedom-e-sdk.html#using-with-freertos

iachievedit, 2020-06-14:
Thanks @valeros I'll update my `platformio.ini` accordingly and give it a shot!

closed 2020-09-16 by stale[bot]