platformio/platformio-home #932: Could not import project

galibimtiaz, 2020-06-27:
PIO Core Call Error: "framework-stm32cube @ 2.0.181130 has been successfully installed!\r\n======================== [SUCCESS] Took 717.73 seconds ========================\r\n\r\nEnvironment         Status    Duration\r\n------------------  --------  ------------\r\nnucleo_f072rb       SUCCESS   00:11:57.731\r\nnucleo_f103rb       IGNORED\r\nnucleo_f207zg       IGNORED\r\nnucleo_f334r8       IGNORED\r\ngenericSTM32F303CB  IGNORED\r\nnucleo_f401re       IGNORED\r\nnucleo_f746zg       IGNORED\r\nnucleo_l053r8       IGNORED\r\nnucleo_l152re       IGNORED\r\nnucleo_l476rg       IGNORED\r\ncloud_jam           IGNORED\r\ncloud jam l4        IGNORED\r\n========================= 1 succeeded in 00:11:57.731 ========================="

valeros, 2020-06-29:
Hi @galibimtiaz ! What was the error exactly? I don't see any issues in the log.

galibimtiaz, 2020-09-11:
I sometimes get this error when I try to download any example sketch, some times it takes almost 10 min to download an example sketch and sometimes it gets hang.

closed 2020-10-30 by ivankravets