platformio/platformio-home #936: Could not import Arduino project

qzzf1w, 2020-06-28:
PIO Core Call Error: "The current working directory\n C:\\Users\\qzzf1w\\Documents\\PlatformIO\\Projects\\200628-140115-teensylc \nwill be used for the project.\n\nThe next files/directories have been created in \u001b[36mC:\\Users\\qzzf1w\\Documents\\PlatformIO\\Projects\\200628-140115-teensylc\u001b[0m\n\u001b[36minclude\u001b[0m - Put project header files here\n\u001b[36mlib\u001b[0m - Put here project specific (private) libraries\n\u001b[36msrc\u001b[0m - Put project source files here\n\u001b[36mplatformio.ini\u001b[0m - Project Configuration File\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 134, in project_init\n    init_base_project(project_dir)\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 174, in init_base_project\n    config.save()\n  File \"c:\\users\\qzzf1w\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\config.py\", line 463, in save\n    with open(path or self.path, \"w+\") as fp:\nIOError: [Errno 13] Permission denied: 'C:\\\\Users\\\\qzzf1w\\\\Documents\\\\PlatformIO\\\\Projects\\\\200628-140115-teensylc\\\\platformio.ini'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

qzzf1w, 2020-07-03:
Still gettking this. Have upgraed

valeros, 2020-07-03:
Maybe it's blocking an antivirus  PlatformIO? Or you don't have rights to write to `C:\\Users\\qzzf1w\\Documents\\PlatformIO\\Projects\\200628-140115-teensylc` folder? Could you please try to create project in a separate folder, somewhere in your system?

closed 2020-10-30 by ivankravets