platformio/platformio-home #937: Can't create ESP32 Project

Shadowlauch, 2020-06-30:
I have been trying to get an ESP-32 project up and running and I just can't get it to work. That is the current error I get.

PIO Core Call Error: "The current working directory C:\\Users\\Lars\\Documents\\PlatformIO\\Projects\\200630-075926-espidf-blink will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Lars\\Documents\\PlatformIO\\Projects\\200630-075926-espidf-blink\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\r\nPLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf 3.40001.200521 (4.0.1) \r\n - tool-cmake 3.16.5 \r\n - tool-esptoolpy 1.20600.0 (2.6.0) \r\n - tool-idf 1.0.1 \r\n - tool-mconf 1.4060000.20190628 (406.0.0) \r\n - tool-ninja 1.9.0 \r\n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n - toolchain-xtensa32 2.80200.200226 (8.2.0)\r\nReading CMake configuration...\r\n-- Found Git: C:/Program Files/Git/cmd/git.exe (found version \"2.27.0.windows.1\") \r\n-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32\r\n-- IDF_TARGET not set, using default target: esp32\r\n-- The C compiler identification is GNU 8.2.0\r\n-- The CXX compiler identification is GNU 8.2.0\r\n-- The ASM compiler identification is GNU\r\n-- Found assembler: C:/Users/Lars/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/Lars/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\r\n-- Check for working C compiler: C:/Users/Lars/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works\r\n-- Detecting C compiler ABI info\r\n-- Detecting C compiler ABI info - done\r\n-- Detecting C compile features\r\n-- Detecting C compile features - done\r\n-- Check for working CXX compiler: C:/Users/Lars/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe\r\n-- Check for working CXX compiler: C:/Users/Lars/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works\r\n-- Detecting CXX compiler ABI info\r\n-- Detecting CXX compiler ABI info - done\r\n-- Detecting CXX compile features\r\n-- Detecting CXX compile features - done\r\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\r\n-- Project version: 1\r\n-- Building ESP-IDF components for target esp32\r\n-- Checking Python dependencies...\r\n-- Configuring incomplete, errors occurred!\r\nSee also \"C:/Users/Lars/Documents/PlatformIO/Projects/200630-075926-espidf-blink/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log\".\r\n\r\nfatal: not a git repository (or any of the parent directories): .git\r\nCMake Error at C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):\r\n Some Python dependencies must be installed. Check above message for\r\n details.\r\nCall Stack (most recent call first):\r\n C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)\r\n C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)\r\n CMakeLists.txt:3 (project)\r\n\r\n\r\n\r\n========================== [FAILED] Took 4.06 seconds =========================="

valeros, 2020-06-30:
Hi @Shadowlauch ! It looks like some of the Python dependencies weren't installed properly. What python version is installed in your system? Could you please try to run the following command in the IDE terminal `C:/Users/Lars/penv/Scripts/pip install "cryptography>=2.1.4" "future>=0.15.2" "pyparsing>=2.0.3,<2.4.0"`

Shadowlauch, 2020-06-30:
Hey @valeros thanks for the quick answer. I just updated it yesterday while trying to troubleshoot this error so 3.8.3. When I try the command you posted I get the following response:

```
Requirement already satisfied: cryptography>=2.1.4 in c:\users\lars\.platformio\penv\lib\site-packages (2.9.2)
Requirement already satisfied: future>=0.15.2 in c:\users\lars\.platformio\penv\lib\site-packages (0.18.2)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\lars\.platformio\penv\lib\site-packages (2.3.1)
Requirement already satisfied: six>=1.4.1 in c:\users\lars\.platformio\penv\lib\site-packages (from cryptography>=2.1.4) (1.15.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\lars\.platformio\penv\lib\site-packages (from cryptography>=2.1.4) (1.14.0)
Requirement already satisfied: pycparser in c:\users\lars\.platformio\penv\lib\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4) (2.20)

```

valeros, 2020-06-30:
Could  you please try to compile your project in verbose mode? `pio run -v`

Shadowlauch, 2020-06-30:
```
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.12.4 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.5
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
-- Configuring incomplete, errors occurred!
See also "C:/Users/Lars/Documents/PlatformIO/Projects/BLEPedal/.pio/build/nodemcu-32s/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)
  C:/Users/Lars/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:3 (project)

=============================================================================================== [FAILED] Took 2.31 seconds =============================================================================================== 

```

valeros, 2020-06-30:
The error message is not really helpful, what is the contents of `C:/Users/Lars/Documents/PlatformIO/Projects/BLEPedal/.pio/build/nodemcu-32s/CMakeFiles/CMakeOutput.log`

Shadowlauch, 2020-06-30:
I have attached the file

[CMakeOutput.log](https://github.com/platformio/platformio-home/files/4851854/CMakeOutput.log)

valeros, 2020-06-30:
Is there `CMakeError.log` file near `CMakeOutput.log` ? What Python version is installed in your system?

Shadowlauch, 2020-06-30:
I checked all the files inside C:/Users/Lars/Documents/PlatformIO/Projects/BLEPedal/.pio/build/ and I couldn't find any other log files.

valeros, 2020-06-30:
What Python version is installed in your system?

Shadowlauch, 2020-06-30:
The one that is PATH/PYTHONPATH/PYTHON is `Python 3.8.3`

valeros, 2020-06-30:
And this one `C:/Users/Lars/penv/Scripts/python --version` ?

Shadowlauch, 2020-06-30:
```
PS C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal> C:/Users/Lars/.platformio/penv/Scripts/python --version
Python 3.8.3
```

valeros, 2020-06-30:
That's strange, let's try to manually detect what is wrong. From the terminal window run the following commands:
```
set IDF_PATH=C:/Users/Lars/.platformio/packages/framework-espidf
C:/Users/Lars/.platformio/penv/Scripts/python C:/Users/Lars/.platformio/packages/framework-espidf/tools/check_python_dependencies.py
```
After that, reset the value for `IDF_PATH`:
```
set IDF_PATH=
```

Shadowlauch, 2020-06-30:
Ok so I ran this in the terminal:

```

C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal>set IDF_PATH=C:/Users/Lars/.platformio/packages/framework-espidf

C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal>C:/Users/Lars/.platformio/penv/Scripts/python C:/Users/Lars/.platformio/packages/framework-espidf/tools/check_python_dependencies.py
Python requirements from C:/Users/Lars/.platformio/packages/framework-espidf\requirements.txt are satisfied.

C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal>
C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal>set IDF_PATH=

C:\Users\Lars\Documents\PlatformIO\Projects\BLEPedal>
```

valeros, 2020-06-30:
As you can see that script has no complaints, so probably you'll need to do some "print" debug in `C:/Users/Lars/.platformio/packages/framework-espidf/tools/check_python_dependencies.py`. For example, you can add 
```
print("Not satisfied packages:", not_satisfied)
```
above line 63, then clean your PIO project and try to compile it again.

Shadowlauch, 2020-06-30:
@valeros Thank you for pointing me in the right direction. It was 2 problems at once:

1.  The PYTHON env variable did not end with "python.exe", but "python" that could not be executed
2. The IDF_PATH env variable was not set 

in both cases the python script did exit on an error and halt the whole process.

closed 2020-06-30 by Shadowlauch