platformio/platformio-home #940: Could not load recent projects

exothink, 2020-06-30:
Server error: {"args":[2,"No such file or directory"],"message":"[Errno 2] No such file or directory: 'C:\\\\Users\\\\EliteDesk\\\\.platformio\\\\platforms\\\\Arduino_Core_STM32\\\\platform.json'","type":"FileNotFoundError"}

I can't load Platform... etc.  tried deleting 'penv' folder. No help.

valeros, 2020-06-30:
`C:\\Users\\EliteDesk\\.platformio\\platforms\\Arduino_Core_STM32` what is this folder? How did it get in PlatformIO Home directory? Try to delete it.

closed 2020-06-30 by valeros

exothink, 2020-06-30:
I have it working now.  

At one time I hacked an install of STM23Arduino 1.9 before it was available on PIO.  I deleted the PIO extension and reinstalled it. All is well.  I also uninstalled all old STM23Arduino versions.