platformio/platformio-home #956: Error occurred while requesting content from https://dl.registry.nm1.platformio.org/examples/26/4c/952336c9a92b81be05202ed0a03453c44d390f4062b2af61c8a87bb71d64/StallGuard_TMC2660.ino

mralitsch, 2020-07-05:
HTTPSConnectionPool(host='dl.registry.nm1.platformio.org', port=443): Max retries exceeded with url: /examples/26/4c/952336c9a92b81be05202ed0a03453c44d390f4062b2af61c8a87bb71d64/StallGuard_TMC2660.ino (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')))

mralitsch, 2020-07-05:
PS D:\Milan Documents\3D printing\Marlin\Marlin 2.0.5.3 VC>  platformio lib install 5513
Library Storage: D:\Milan Documents\3D printing\Marlin\Marlin 2.0.5.3 VC\.pio\libdeps\mega2560
LibraryManager: Installing id=5513
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 381, in _make_request
    self._validate_conn(conn)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 976, in _validate_conn
    conn.connect()
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\connection.py", line 370, in connect
    ssl_context=context,
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 377, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "C:\Users\mrali\.platformio\python37\lib\ssl.py", line 423, in wrap_socket
    session=session
  File "C:\Users\mrali\.platformio\python37\lib\ssl.py", line 870, in _create
    self.do_handshake()
  File "C:\Users\mrali\.platformio\python37\lib\ssl.py", line 1139, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 725, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\urllib3\util\retry.py", line 439, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='dl.registry.nm1.platformio.org', port=443): Max retries exceeded with url: /libraries/b7/0e/863b35a714c87e5862a75e54bb848223feabc4fc5f443438a90d0c2da0c3/TMCStepper-0.7.1.tar.gz (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 155, in lib_install
    library, silent=silent, interactive=interactive, force=force
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 281, in install
    force=force,
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 704, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 131, in _install_from_piorepo
    requirements,
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 480, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 195, in download
    raise e
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 179, in download
    fd = FileDownloader(url, dest_dir)
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\platformio\downloader.py", line 42, in __init__
[... 9 lines cut ...]
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\requests\sessions.py", line 665, in <listcomp>
    history = [resp for resp in gen]
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\requests\sessions.py", line 245, in resolve_redirects
    **adapter_kwargs
  File "C:\Users\mrali\.platformio\penv\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
[... cut ...]
unable to get local issuer certificate (_ssl.c:1076)')))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2020-09-09:
Resolved in the latest release.

closed 2020-09-09 by ivankravets