platformio/platformio-home #963: Could not import project

teslawei, 2020-07-06:
PIO Core Call Error: "The current working directory /Users/tesla/Documents/PlatformIO/Projects/200705-225744-zephyr-blink will be used for the project.\n\nThe next files/directories have been created in /Users/tesla/Documents/PlatformIO/Projects/200705-225744-zephyr-blink\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing adafruit_feather_nrf52832 (platform: nordicnrf52; framework: zephyr; board: adafruit_feather_nrf52832)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html\nPLATFORM: Nordic nRF52 4.2.1 > Adafruit Bluefruit nRF52832 Feather\nHARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash\nDEBUG: Current (jlink) External (jlink, stlink)\nPACKAGES: \n - framework-zephyr 2.20200.200401 (2.2.0) \n - framework-zephyr-canopennode 0.20200.200115 (2.2.0) \n - framework-zephyr-civetweb 0.20200.190807 (2.2.0) \n - framework-zephyr-fatfs 0.20200.191125 (2.2.0) \n - framework-zephyr-hal-nordic 0.20200.200130 (2.2.0) \n - framework-zephyr-libmetal 0.20200.190530 (2.2.0) \n - framework-zephyr-littlefs 0.20200.190811 (2.2.0) \n - framework-zephyr-loramac-node 0.20200.191220 (2.2.0) \n - framework-zephyr-lvgl 0.20200.191207 (2.2.0) \n - framework-zephyr-mbedtls 0.20200.200206 (2.2.0) \n - framework-zephyr-mcuboot 0.20200.200219 (2.2.0) \n - framework-zephyr-mcumgr 0.20200.200117 (2.2.0) \n - framework-zephyr-mipi-sys-t 0.20200.191024 (2.2.0) \n - framework-zephyr-nrf-hw-models 0.20200.191115 (2.2.0) \n - framework-zephyr-open-amp 0.20200.190612 (2.2.0) \n - framework-zephyr-openthread 0.20200.191014 (2.2.0) \n - framework-zephyr-segger 0.20200.190421 (2.2.0) \n - framework-zephyr-tinycbor 0.20200.191016 (2.2.0) \n - tool-cmake 3.16.4 \n - tool-dtc 1.4.7 \n - tool-gperf 3.1.0 \n - tool-ninja 1.9.0 \n - tool-sreccat 1.164.0 (1.64) \n - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)\nPython 3.4 or greater required, but you have Python 2.7.15\n========================== [FAILED] Took 1.89 seconds ==========================\n\nEnvironment                Status    Duration\n-------------------------  --------  ------------\nadafruit_feather_nrf52832  FAILED    00:00:01.886\nnrf52_dk                   IGNORED\nnrf52840_dk                IGNORED\nparticle_xenon             IGNORED\n==================== 1 failed, 0 succeeded in 00:00:01.886 ===================="

teslawei, 2020-07-06:
I have python 3.7 installed on my machine, but got this error.

valeros, 2020-07-06:
Hi @teslawei ! Looks like your PlatformIO virtual environment was created for Python 2.7.
- Close the IDE
- Make sure Python 3.x is the main Python installation in your system.
- Delete folder `C:\Users\Ali\.platformio\.penv`
- Open the IDE and wait until a new installation is completed

closed 2020-07-06 by valeros

teslawei, 2020-07-06:
Hi Valeros,

I followed the instruction, but python3 couldn't be installed to .penv.
I have the python 3 as my main python in my OS.
Is there any other way to install python3 to penv?

Thanks

valeros, 2020-07-07:
> I followed the instruction, but python3 couldn't be installed to .penv.

What do you mean? Were there any errors?

teslawei, 2020-07-10:
I mean in the penv folder, PIO couldn't find python3, but only python 2.

valeros, 2020-07-13:
So Python2 is set as the main Python in your system. I'm not sure how Python is installed in your system, but if you specify path to Python 3 to be first in the PATH environment variable, then PlatformIO will be able to find it.