platformio/platformio-home #965: Libraries: Could not update library

danilius, 2020-07-06:
PIO Core Call Error: "Library Storage: F:\\Work2\\Albrecht Panknin\\code\\OV7670-ESP32\\.pio\\libdeps\\esp32dev\r\nUpdating ov7670                          @ 119b4c04a4     [0.0.0-alpha+sha.119b4c04a4]\r\nUninstalling ov7670 @ 119b4c04a4: \t[OK]\r\nLooking for ov7670 library in registry\r\nov7670\r\n======\r\n#ID: 2606\r\nOV7670 Library\r\n\r\nKeywords: camera, ov7670\r\nCompatible frameworks: Mbed\r\nCompatible platforms: Atmel SAM, Freescale Kinetis, Nordic nRF51, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, Nordic nRF52, Maxim 32, WIZNet W7500, RISC-V GAP, NXP i.MX RT\r\nAuthors: Edoardo De Marchi\r\n\r\nOV7670\r\n======\r\n#ID: 4985\r\nTest Acquis Cam2D\r\n\r\nKeywords: ov7670\r\nCompatible frameworks: Mbed\r\nCompatible platforms: Atmel SAM, Freescale Kinetis, Nordic nRF51, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, Nordic nRF52, Maxim 32, WIZNet W7500, RISC-V GAP, NXP i.MX RT\r\nAuthors: The DarkRunners\r\n\r\nFound: https://platformio.org/lib/show/2606/ov7670\r\nLibraryManager: Installing id=2606 @ 0.0.0-alpha+sha.119b4c04a4\r\n\n\nConflict: More than one library has been found by request {\"name\": \"ov7670\", \"requirements\": \"0.0.0-alpha+sha.119b4c04a4\"}:\r\nAutomatically chose the first available library (use `--interactive` option to make a choice)\r\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 540, in _install_from_tmp_dir\r\n    ), mismatch_error\r\nAssertionError: Package version 119b4c04a4 doesn't satisfy requirements 0.0.0-alpha+sha.119b4c04a4\r\n\r\nDuring handling of the above exception, another exception occurred:\r\n\r\nTraceback (most recent call last):\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 260, in lib_update\r\n    lm.update(library, only_check=only_check)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 810, in update\r\n    self.install(name, latest, after_update=True)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 281, in install\r\n    force=force,\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 704, in install\r\n    pkg_dir = self._install_from_piorepo(name, requirements)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 131, in _install_from_piorepo\r\n    requirements,\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 499, in _install_from_url\r\n    return self._install_from_tmp_dir(_tmp_dir, requirements)\r\n  File \"c:\\users\\plane\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 542, in _install_from_tmp_dir\r\n    assert tmp_manifest[\"version\"] == requirements, mismatch_error\r\nAssertionError: Package version 119b4c04a4 doesn't satisfy requirements 0.0.0-alpha+sha.119b4c04a4\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-07-09:
Sorry for the issue. Fixed. Please try again.

closed 2020-07-09 by ivankravets

danilius, 2020-07-09:
Awesome, thanks.