platformio/platformio-home #970: Could not load installed platforms

JBHop, 2020-07-07:
PIO Core Call Error: "Error: Unknown development platform 'C:\\Users\\Brian\\.platformio\\platforms\\ststm32'"

valeros, 2020-07-07:
Hi @JBHop ! Looks like the platform in `C:\Users\Brian\.platformio\platforms\ststm32` folder is broken in some way. Try to delete `C:\Users\Brian\.platformio\platforms\ststm32` and install again.

JBHop, 2020-07-08:
Yes the directory was broken!  The problem started when I did a " pio lib update".
When the update reached the Framework  STM32 section, it went to replace the STM32 library version 7.0.0 with 7.1.0. It failed, then tried a different mirror, which also failed.
A second attempt at "pio lib update" resulted in the same errors. Looked like the install package for 7.1.0 was inserting far too many directory \\\\\'s. 
Rolled STM32 back to 7.0.0, until this morning, when version 7.1.1 successfully installed ! Yeah!
Now, if all all the other problems with my build will go away...
Thanks for the reply and support!

closed 2020-07-08 by JBHop