platformio/platformio-home #1013: Could not load boards list

ThinCode2020, 2020-07-18:
PIO Core Call Error: "Error: Please specify name, url and vendor fields for C:\\Users\\Louis\\.platformio\\platforms\\atmelmegaavr\\boards\\package_thinary_index.json".

I am trying to create a new project, as soon as I open PIO and click new project. It gave an error above.

Before that, I copy a json file "package_thinary_index.json" into "C:\\Users\\Louis\\.platformio\\platforms\\atmelmegaavr\\boards\\".
I don't know how to install a new board. The Thinary Nano Every (Atmega4808) is my new board and I want to test it in PlatformIO.

I appreciate for any help.

Thank you,

Louis

ThinCode2020, 2020-07-18:
Here is the "package_thinary_index.json" file:

`{
  "packages":[
    {
      "name":"thinary",
      "maintainer":"Thinary Electronic",
      "websiteURL":"http://thinary.cn",
      "email":"thinary@163.com",
      "help":{
      "online":"http://thinary.cn"
      },
      "platforms": [
        {
          "name": "Thinary AVR Boards",
          "architecture": "avr",
          "version": "1.0.0",
          "category": "Thinary",
          "url": "https://github.com/Thinary/ThinaryArduino/releases/download/1.0.0/thinary-megaavr-1.0.0.zip",
          "archiveFileName": "thinary-megaavr-1.0.0.zip",
          "checksum": "SHA-256:cc9b7471b970794b1c2bf04724431a6425a83cd11b6049a9ab61af961f5cf5f3",
          "size": "990906",
          "help": {
            "online": "https://github.com/Thinary/ThinaryArduino/issues"
          },
          "boards": [
            {
              "name": "Thinary Nano Every(ATmega4808)"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino16"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
        }
      ],
      "tools":[
        {
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-arduino5",
          "systems": [
            {
              "size": "34462042",
              "checksum": "SHA-256:f4acd5531c6b82c715e2edfa0aadb13fb718b4095b3ea1aa1f7fbde680069639",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-arm-linux-gnueabihf.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-arm-linux-gnueabihf.tar.bz2"
            },
            {
              "size": "39381972",
              "checksum": "SHA-256:dd9c70190be370a44fb47dab1514de6d8852b861dfa527964b65c740d8d50c10",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "38492678",
[... 104 lines cut ...]
              "host": "aarch64-linux-gnu",
              "archiveFileName": "arduinoOTA-1.3.0-linux_arm64.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/arduinoOTA-1.3.0-linux_arm64.tar.bz2"
            },
            {
              "size": "2766116",
              "checksum": "SHA-256:d5d0f82ff829c0e434d12a2ee640a6fbd78f893ab37782edbb8b5bf2359d119e",
              "host": "i386-apple-darwin11",
              "archiveFileName": "arduinoOTA-1.3.0-darwin_amd64.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/arduinoOTA-1.3.0-darwin_amd64.tar.bz2"
            },
            {
              "size": "2768948",
              "checksum": "SHA-256:051943844eee442460d2c709edefadca184287fffd2b6c100dd53aa742aa05f6",
              "host": "i686-mingw32",
              "archiveFileName": "arduinoOTA-1.3.0-windows_386.zip",
              "url": "http://downloads.arduino.cc/tools/arduinoOTA-1.3.0-windows_386.zip"
            }
          ]
        }
        ]
      }
   ]
}
`

valeros, 2020-07-20:
Hi @ThinCode2020 ! You need to delete the file `C:\Users\Louis\.platformio\platforms\atmelmegaavr\boards\package_thinary_index.json` it shouldn't be there since `boards` folder must contain only valid board manifests.

closed 2020-07-20 by valeros