platformio/platformio-home #1020: Could not load boards list

kevno08, 2020-07-21:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 31, in cli\r\n    return _print_boards_json(query, installed)\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 80, in _print_boards_json\r\n    for board in _get_boards(installed):\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 75, in _get_boards\r\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 208, in get_all_boards\r\n    for board in self.get_registered_boards():\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 202, in get_registered_boards\r\n    return util.get_api_result(\"/boards\", cache_valid=\"7d\")\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 336, in get_api_result\r\n    with ContentCache() as cc:\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\app.py\", line 182, in __enter__\r\n    self.delete()\r\n  File \"c:\\users\\kevorknormanian\\.platformio\\penv\\lib\\site-packages\\platformio\\app.py\", line 295, in delete\r\n    with open(self._db_path, \"w\") as fp:\r\nPermissionError: [Errno 13] Permission denied: 'C:\\\\Users\\\\KevorkNormanian\\\\.platformio\\\\.cache\\\\db.data'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

kevno08, 2020-07-21:
Why am I getting this message?

valeros, 2020-07-21:
When it happens? It looks like your user doesn't have the necessary privileges to perform this action.

kevno08, 2020-07-21:
when I try to open project in Vs code
and when I build it says permission denied.

valeros, 2020-07-21:
Is there an antivirus in your system that can be blocking PlatformIO? Try chaning [`core_dir`](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#core-dir) option to a folder your user can write to.

kevno08, 2020-07-21:
where to go to change the core_dir? Don't see it in the PlatformIO.ini

valeros, 2020-07-21:
Create a new project manually, then specify the `core_dir` in `platformio.ini` file, something like this:
```
[platformio]
core_dir = YOUR_PATH, e.g. C:\\.platformio

[env:your_env]
platform = ...
framework = ...
board = ...
```

kevno08, 2020-07-21:
But I try to open a Marlin in the VS and get this code as well. Can I create a new project and then import marlin into that project?

valeros, 2020-07-21:
You can create a new project and then copy marlin files to that new project.

closed 2020-09-19 by stale[bot]