platformio/platformio-home #1048: Could not import Arduino project

bhsong1011, 2020-07-31:
PIO Core Call Error: "The current working directory C:\\Users\\송병현\\Documents\\PlatformIO\\Projects\\200731-100209-megaatmega2560 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\송병현\\Documents\\PlatformIO\\Projects\\200731-100209-megaatmega2560\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html\r\nPLATFORM: Atmel AVR 2.2.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)\r\nHARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash\r\nDEBUG: Current (simavr) On-board (simavr)\r\nPACKAGES: \r\n - framework-arduino-avr 5.0.0 \r\n - toolchain-atmelavr 1.50400.190710 (5.4.0)\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nFound 5 compatible libraries\r\nScanning dependencies...\r\nNo dependencies\r\nBuilding in release mode\r\nException in thread Thread-6:\r\nTraceback (most recent call last):\r\n  File \"C:\\Python\\Python38-32\\lib\\threading.py\", line 932, in _bootstrap_inner\r\n    self.run()\r\n  File \"C:\\Python\\Python38-32\\lib\\threading.py\", line 870, in run\r\n    self._target(*self._args, **self._kwargs)\r\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 48, in run\r\n    self.do_reading()\r\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 101, in do_reading\r\n    for line in iter(self._pipe_reader.readline, \"\"):\r\nUnicodeDecodeError: 'cp949' codec can't decode byte 0xec in position 756: illegal multibyte sequence\r\n========================= [SUCCESS] Took 0.65 seconds ========================="

In proc.py line 30, class AsyncPipeBase(object) has fdopen which occurs encoding error. 

`class AsyncPipeBase(object):
    def __init__(self):
        self._fd_read, self._fd_write = os.pipe()
        self._pipe_reader = os.fdopen(self._fd_read, 'rt', encoding='UTF8')
        # self._pipe_reader = os.fdopen(self._fd_read)
        self._buffer = ""
        self._thread = Thread(target=self.run)
        self._thread.start()
`
If you change the fdopen with encoding in UTF8, the problem disappears. However, I still have unicode error line in the VSCode terminal when I build a .ino file. 

> Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\송병현\Documents\PlatformIO\Projects\200731-100001-megaatmega2560\src\test.ino.cpp file or convert it to UTF-8

Please take a look in this error.

ivankravets, 2020-09-09:
This issue is stale because it has been open 30 days with no activity. Please provide more details or this will be closed in 5 days

bhsong1011, 2020-09-10:
I'm on Windows with my account name in Korean, which caused fatal errors for PlatformIO.
Because python wasn't able to read Korean in the file path for some reason, probably related to encoding, the whole program kept failing.

When I first installed VSCode&PlatformIO in my C:\Users\(Korean), the installation itself failed due to that error.
After I tried installing both to the D:\ where there were no exceptional characters, the installation went fine but the build function was the next problem which was the context of the original issue.

I fixed the error by adding utf-8 encoding option in fdopen( ) at proc.py.

Can PlatformIo have try-except code with all file opening functions added with encoding options, for those aren't using English as their first languages? Or at least please warn users that Non ASCII characters in the path will cause fatal errors in the programs therefore they should mind the installation&work paths?

Thanks.

ivankravets, 2020-09-13:
Could you upgrade to the latest PlatformIO IDE 2.0 for the VSCode? Does it work now?

bhsong1011, 2020-09-16:
I tried installing PlatformIO on another desktop today, but it failed.

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\.platformio\.cache\tmp\.piocore-installer-_0w54rv2\tmpzqdyz3z9\pioinstaller.zip\pioinstaller\python.py", line 174, in find_compatible_pythons
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbc in position 27: invalid start byte
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\.platformio\python3\python.exe
Creating a virtual environment at C:\.platformio\penv
.....

Still there is the encoding error.

ivankravets, 2020-09-16:
Fixed in https://github.com/platformio/platformio-core-installer/issues/56

closed 2020-09-16 by ivankravets