platformio/platformio-home #1052: Could not import project

OzHawk, 2020-07-31:
PIO Core Call Error: "framework-mbed @ 6.60200.200722 has been successfully installed!\r\nCorePackageManager: Installing tool-scons @ ~3.30102.0\r\ntool-scons @ 3.30102.0 has been successfully installed!\r\n======================== [SUCCESS] Took 496.27 seconds ========================\r\n\r\nEnvironment    Status    Duration\r\n-------------  --------  ------------\r\nnucleo_f412zg  SUCCESS   00:08:16.272\r\nnucleo_l4r5zi  IGNORED\r\n========================= 1 succeeded in 00:08:16.272 ========================="'

It reported both an Error and Success at the same time

ivankravets, 2020-09-09:
This issue is stale because it has been open 30 days with no activity. Please provide more details or this will be closed in 5 days

OzHawk, 2020-09-10:
I'm not sure how much more information you need.  As you can see from the log, under the title "PIO Core Call Error" the program reported the "SUCCESS" of the action.
Is it an error?  It seems like where this log entry is initiated from is calling the wrong method, or passing the wrong parameters.  I would expect it to be "PIO Core Call Info" or something similar, as it clearly is not an error and adds to user confusion.

ivankravets, 2020-09-12:
Do you use the latest PlatformIO IDE 2.0.1?

OzHawk, 2020-09-12:
I'm using PlatformIO within Visual Studio Code
Visual Studio Code Version:
August 2020 (version 1.49)

PlatformIO Core Version:
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.

I wasn't on PlatformIDE 2.0.1, I would have been on the previous latest release at the time that the issue was logged.  2.0.1 only came out on the 10 September.

closed 2020-09-19 by stale[bot]