platformio/platformio-home #1058: Could not initialize project

jacumol, 2020-08-03:
PIO Core Call Error: "The current working directory /Users/jacumol/Pitmmam/Repositorios/tests/Test-WTD will be used for the project.\n\nThe next files/directories have been created in /Users/jacumol/Pitmmam/Repositorios/tests/Test-WTD\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing wemos_d1_mini32 (platform: espressif32; board: wemos_d1_mini32; framework: arduino)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nError: Unknown board ID 'wemos_d1_mini32'\n========================== [FAILED] Took 1.40 seconds =========================="

valeros, 2020-08-03:
Hi @jacumol ! What version of the `espressif32` platform do you use?

ivankravets, 2020-09-09:
Please update all dev-platforms and restart VSCode.

closed 2020-09-09 by ivankravets