platformio/platformio-home #1073: Libraries: Data

ijash, 2020-08-07:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/platformio/commands/lib.py\", line 416, in lib_show\n    lib = util.get_api_result(\"/lib/info/%d\" % lib_id, cache_valid=\"1d\")\n  File \"/home/ijash/.platformio/penv/lib/python3.6/site-packages/platformio/util.py\", line 340, in get_api_result\n    return json.loads(result)\n  File \"/usr/lib/python3.6/json/__init__.py\", line 354, in loads\n    return _default_decoder.decode(s)\n  File \"/usr/lib/python3.6/json/decoder.py\", line 339, in decode\n    obj, end = self.raw_decode(s, idx=_w(s, 0).end())\n  File \"/usr/lib/python3.6/json/decoder.py\", line 357, in raw_decode\n    raise JSONDecodeError(\"Expecting value\", s, err.value) from None\njson.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2020-08-10:
Hi @ijash ! What library you were trying to install when this error occurred? Looks like that library has a broken manifest.

ijash, 2020-08-10:
It's DHT library

valeros, 2020-08-10:
Unfortunately, I cannot reproduce the issue. Feel free to reopen if the error happens again.

closed 2020-08-10 by valeros