platformio/platformio-home #1077: Could not initialize project

pbillmann, 2020-08-09:
PIO Core Call Error: "The current working directory /Users/amandabillmann/Documents/PlatformIO/Projects/cam2 will be used for the project.\n\nThe next files/directories have been created in /Users/amandabillmann/Documents/PlatformIO/Projects/cam2\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32cam (platform: espressif32; board: esp32cam; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html\nPLATFORM: Espressif 32 1.12.4 #08ef4bf > AI Thinker ESP32-CAM\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40001.200521 (4.0.1) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.190618 (2.28.51) \n - toolchain-xtensa32 2.80200.200226 (8.2.0)\n/Users/amandabillmann/.platformio/penv/lib/python2.7/site-packages/cryptography/__init__.py:39: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.\n  CryptographyDeprecationWarning,\nReading CMake configuration...\n-- Found Git: /usr/local/bin/git (found version \"2.28.0\") \n-- IDF_TARGET not set, using default target: esp32\n-- The C compiler identification is GNU 8.2.0\n-- The CXX compiler identification is GNU 8.2.0\n-- The ASM compiler identification is GNU\n-- Found assembler: /Users/amandabillmann/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/amandabillmann/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/amandabillmann/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc -- works\n-- Detecting C compiler ABI info\n-- Detecting C compiler ABI info - done\n-- Detecting C compile features\n-- Detecting C compile features - done\n-- Check for working CXX compiler: /Users/amandabillmann/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++\n-- Check for working CXX compiler: /Users/amandabillmann/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++ -- works\n-- Detecting CXX compiler ABI info\n-- Detecting CXX compiler ABI info - done\n-- Detecting CXX compile features\n-- Detecting CXX compile features - done\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\n-- Project version: 1\n-- Building ESP-IDF components for target esp32\n-- Checking Python dependencies...\nThe following Python requirements are not satisfied:\npyparsing>=2.0.3,<2.4.0\nPlease refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.\nAlternatively, you can run \"/Users/amandabillmann/.platformio/penv/bin/python -m pip install -r /Users/amandabillmann/.platformio/packages/framework-espidf/requirements.txt\" for resolving the issue.\n-- Configuring incomplete, errors occurred!\nSee also \"/Users/amandabillmann/Documents/PlatformIO/Projects/cam2/.pio/build/esp32cam/CMakeFiles/CMakeOutput.log\".\nfatal: not a git repository (or any of the parent directories): .git\nCMake Error at /Users/amandabillmann/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):\n  Some Python dependencies must be installed.  Check above message for\n  details.\nCall Stack (most recent call first):\n  /Users/amandabillmann/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)\n  /Users/amandabillmann/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)\n  CMakeLists.txt:3 (project)\n========================== [FAILED] Took 6.57 seconds =========================="

valeros, 2020-08-10:
Hi @pbillmann ! It looks like the Python dependencies weren't installed properly. You can try to run the following command in the IDE terminal:
```bash
/Users/amandabillmann/.platformio/penv/bin/python -m pip install -r /Users/amandabillmann/.platformio/packages/framework-espidf/requirements.txt
```

ivankravets, 2020-09-09:
This issue is stale because it has been open 30 days with no activity. Please provide more details or this will be closed in 5 days

pbillmann, 2020-09-09:
Close it problem was solved with instaling dependencies as indicated

closed 2020-09-09 by ivankravets