platformio/platformio-home #1111: Could not initialize project

mark-FCS, 2020-08-18:
PIO Core Call Error: "The current working directory /Users/markc/Documents/PlatformIO/Projects/BL654 Laser will be used for the project.\n\nThe next files/directories have been created in /Users/markc/Documents/PlatformIO/Projects/BL654 Laser\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing laird_bl654_dvk (platform: nordicnrf52; board: laird_bl654_dvk; framework: zephyr)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/laird_bl654_dvk.html\nPLATFORM: Nordic nRF52 4.4.0 > BL654 Development Kit\nHARDWARE: NRF52840 64MHz, 256KB RAM, 1MB Flash\nDEBUG: Current (jlink) On-board (cmsis-dap, jlink) External (blackmagic, stlink)\nPACKAGES: \n - framework-zephyr 2.20300.200623 (2.3.0) \n - framework-zephyr-canopennode 0.0.0-alpha+sha.5c6b0566d5 \n - framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc \n - framework-zephyr-cmsis 0.0.0-alpha+sha.542b2296e6 \n - framework-zephyr-fatfs 0.0.0-alpha+sha.9ee6b9b951 \n - framework-zephyr-hal-nordic 0.0.0-alpha+sha.742d8fb839 \n - framework-zephyr-libmetal 0.0.0-alpha+sha.3c3c9ec83b \n - framework-zephyr-littlefs 0.0.0-alpha+sha.0aefdda69d \n - framework-zephyr-loramac-node 0.0.0-alpha+sha.29e516ec58 \n - framework-zephyr-lvgl 0.0.0-alpha+sha.74fc2e753a \n - framework-zephyr-mbedtls 0.0.0-alpha+sha.4bf099f125 \n - framework-zephyr-mcuboot 0.0.0-alpha+sha.e88113bbeb \n - framework-zephyr-mcumgr 0.0.0-alpha+sha.5885efb7ca \n - framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.957d46bc3c \n - framework-zephyr-open-amp 0.0.0-alpha+sha.724f7e2a45 \n - framework-zephyr-openthread 0.0.0-alpha+sha.a83d18cf18 \n - framework-zephyr-segger 0.0.0-alpha+sha.6fcf61606d \n - framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4 \n - framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267 \n - framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.7de2daa196 \n - tool-cmake 3.16.4 \n - tool-dtc 1.4.7 \n - tool-gperf 3.1.0 \n - tool-ninja 1.9.0 \n - tool-sreccat 1.164.0 (1.64) \n - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)\nReading CMake configuration...\n-- Application: /Users/markc/Documents/PlatformIO/Projects/BL654 Laser/zephyr\n-- Zephyr version: 2.3.0 (/Users/markc/.platformio/packages/framework-zephyr)\n-- Found Python3: /Users/markc/.platformio/penv/bin/python3 (found suitable exact version \"3.8.5\") found components: Interpreter \n-- Board: bl654_dvk\n-- Found dtc: /Users/markc/.platformio/packages/tool-dtc/dtc (found suitable version \"1.4.7\", minimum required is \"1.4.6\")\n-- Found toolchain: gnuarmemb (/Users/markc/.platformio/packages/toolchain-gccarmnoneeabi)\n-- Found BOARD.dts: /Users/markc/.platformio/packages/framework-zephyr/boards/arm/bl654_dvk/bl654_dvk.dts\n-- Generated zephyr.dts: /Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/zephyr/zephyr.dts\n-- Generated devicetree_unfixed.h: /Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/zephyr/include/generated/devicetree_unfixed.h\nParsing /Users/markc/.platformio/packages/framework-zephyr/Kconfig\nLoaded configuration '/Users/markc/.platformio/packages/framework-zephyr/boards/arm/bl654_dvk/bl654_dvk_defconfig'\nConfiguration saved to '/Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/zephyr/.config'\nKconfig header saved to '/Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/zephyr/include/generated/autoconf.h'\n-- The C compiler identification is GNU 8.2.1\n-- The CXX compiler identification is GNU 8.2.1\n-- The ASM compiler identification is GNU\n-- Found assembler: /Users/markc/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc\n-- Cache files will be written to: /Users/markc/Library/Caches/zephyr\n-- Configuring incomplete, errors occurred!\nSee also \"/Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/CMakeFiles/CMakeOutput.log\".\nSee also \"/Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/CMakeFiles/CMakeError.log\".\n\nPython path: /Users/markc/.platformio/penv/bin/python3\nCMake Error: Could not find cmake module file: CMakeDetermineLaserCompiler.cmake\nCMake Error: Error required internal CMake variable not set, cmake may not be built correctly.\nMissing variable is:\nCMAKE_Laser_COMPILER_ENV_VAR\nCMake Error: Error required internal CMake variable not set, cmake may not be built correctly.\nMissing variable is:\nCMAKE_Laser_COMPILER\nCMake Error: Could not find cmake module file: /Users/markc/Documents/PlatformIO/Projects/BL654 Laser/.pio/build/laird_bl654_dvk/CMakeFiles/3.16.4/CMakeLaserCompiler.cmake\nCMake Error at CMakeLists.txt:3 (project):\n  No CMAKE_Laser_COMPILER could be found.\n\n  Tell CMake where to find the compiler by setting the CMake cache entry\n  CMAKE_Laser_COMPILER to the full path to the compiler, or to the compiler\n  name if it is in the PATH.\n\n\nCMake Error: Could not find cmake module file: CMakeLaserInformation.cmake\nCMake Error: CMAKE_Laser_COMPILER not set, after EnableLanguage\n========================== [FAILED] Took 6.92 seconds =========================="

valeros, 2020-08-18:
Hi @mark-FCS ! That's really strange, do you have any global CMake configurations in your system? What's `CMAKE_Laser_COMPILER`?

mark-FCS, 2020-08-18:
No CMAKE_Laser_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_Laser_COMPILER to the full path to the compiler, or to the compiler
  name if it is in the PATH.

Formatted the error above for better readability.

Hi @valeros thanks for commenting. I'm a little new to VSCode and PlatformIO. I'm not sure where I can check the CMake global configurations. I followed the steps on Zephyr "Getting Started" in order to clone that repo and build a sample app. Then I got VSCode and PlatformIO going with a target board of BL654_DVK, but can't build due to this error.

So I guess I am trying to use the Zephyr CMake package, but I don't have PlatformIO pointed to that correctly. Would that be right? If so, any suggestions? Thanks.

valeros, 2020-08-18:
Could you try to create a new project without whitespace characters in its name?

mark-FCS, 2020-08-19:
@valeros you nailed it. Re-created the project and kept no names in the spaces and everything seems dandy now. Cheers!

closed 2020-08-19 by mark-FCS