platformio/platformio-home #1161: Could not initialize project

Fliptron, 2020-08-29:
PIO Core Call Error: "framework-stm32cube @ 2.0.200813 has been successfully installed!\r\nPackageManager: Installing tool-ldscripts-ststm32 @ ~0.1.0\r\ntool-ldscripts-ststm32 @ 0.1.0 has been successfully installed!\r\n======================== [SUCCESS] Took 183.62 seconds ========================"

valeros, 2020-08-31:
Hi @Fliptron! Unfortunately, the log you provided doesn't contain enough information to help you. Please provide more information about the issue.

closed 2020-08-31 by valeros

Fliptron, 2020-08-31:
Unfortunately, the only additional info I can provide is that I was installing the above mentioned package, and although it reports success, it also popped up an error box, with a submit button to report an error. So I clicked the submit button. I think I was installing for STM32F303CB generic on a Window 7 Pro x64 system.

valeros, 2020-08-31:
Are you able to create a new project or the problem still persists?