platformio/platformio-home #1168: Could not initialize project

burke54, 2020-09-01:
PIO Core Call Error: "The current working directory /freenas/linux/esp32/PlatformIO/Projects/espidf-get-started will be used for the project.\n\nThe next files/directories have been created in /freenas/linux/esp32/PlatformIO/Projects/espidf-get-started\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nPlatformManager: Installing espressif32\nespressif32 @ 1.12.4 has been successfully installed!\nPackageManager: Installing toolchain-xtensa32 @ ~2.50200.0\ntoolchain-xtensa32 @ 2.50200.80 has been successfully installed!\nPackageManager: Installing tool-esptoolpy @ ~1.20600.0\ntool-esptoolpy @ 1.20600.0 has been successfully installed!\nThe platform 'espressif32' has been successfully installed!\nThe rest of packages will be installed automatically depending on your build environment.\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\n--------------------------------------------------------------------------------\nPackageManager: Installing toolchain-xtensa32 @ ~2.80200.0\ntoolchain-xtensa32 @ 2.80200.200827 has been successfully installed!\nPackageManager: Installing toolchain-esp32ulp @ ~1.22851.190618\ntoolchain-esp32ulp @ 1.22851.191205 has been successfully installed!\nPackageManager: Installing framework-espidf @ ~3.40001.0\nframework-espidf @ 3.40001.200521 has been successfully installed!\nPackageManager: Installing tool-cmake @ ~3.16.0\ntool-cmake @ 3.16.4 has been successfully installed!\nPackageManager: Installing tool-ninja @ ^1.7.0\ntool-ninja @ 1.7.1 has been successfully installed!\nCorePackageManager: Installing tool-scons @ ~3.30102.0\ntool-scons @ 3.30102.0 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\nPLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40001.200521 (4.0.1) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-ninja 1.7.1 \n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \n - toolchain-xtensa32 2.80200.200827 (8.2.0)\nInstalling ESP-IDF's Python dependencies\nCollecting cryptography>=2.1.4\n  Using cached cryptography-3.1-cp35-abi3-manylinux2010_x86_64.whl (2.6 MB)\nProcessing /home/burke/.cache/pip/wheels/56/b0/fe/4410d17b32f1f0c3cf54cdfb2bc04d7b4b8f4ae377e2229ba0/future-0.18.2-py3-none-any.whl\nCollecting pyparsing<2.4.0,>=2.0.3\n  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)\nCollecting six>=1.4.1\n  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)\nCollecting cffi!=1.11.3,>=1.8\n  Using cached cffi-1.14.2-cp37-cp37m-manylinux1_x86_64.whl (401 kB)\nCollecting pycparser\n  Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)\nInstalling collected packages: six, pycparser, cffi, cryptography, future, pyparsing\nSuccessfully installed cffi-1.14.2 cryptography-3.1 future-0.18.2 pycparser-2.20 pyparsing-2.3.1 six-1.15.0\nReading CMake configuration...\n-- Found Git: /usr/bin/git (found version \"2.20.1\") \n-- IDF_TARGET not set, using default target: esp32\n-- The C compiler identification is unknown\n-- The CXX compiler identification is unknown\n-- The ASM compiler identification is GNU\n-- Found assembler: /home/burke/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /home/burke/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /home/burke/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc -- broken\n-- Configuring incomplete, errors occurred!\nSee also \"/freenas/linux/esp32/PlatformIO/Projects/espidf-get-started/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log\".\nSee also \"/freenas/linux/esp32/PlatformIO/Projects/espidf-get-started/.pio/build/esp32dev/CMakeFiles/CMakeError.log\".\n\nfatal: not a git repository (or any of the parent directories): .git\nCMake Error at /home/burke/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):\n  The C compiler\n\n    \"/home/burke/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc\"\n\n  is not able to compile a simple test program.\n\n  It fails with the following output:\n\n    Change Dir: /freenas/linux/esp32/PlatformIO/Projects/espidf-get-started/.pio/build/esp32dev/CMakeFiles/CMakeTmp\n    \n    Run Build Command(s):/home/burke/.platformio/packages/tool-ninja/ninja cmTC_c0b36 && [1/2] Building C object CMakeFiles/cmTC_c0b36.dir/testCCompiler.c.obj\n    FAILED: CMakeFiles/cmTC_c0b36.dir/testCCompiler.c.obj \n    /home/burke/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_c0b36.dir/testCCompiler.c.obj   -c testCCompiler.c\n    as: unrecognized option '--longcalls'\n    ninja: build stopped: subcommand failed.\n    \n    \n\n  \n\n  CMake will not be able to correctly generate this project.\nCall Stack (most recent call first):\n  /home/burke/.platformio/packages/framework-espidf/tools/cmake/project.cmake:269 (__project)\n  CMakeLists.txt:3 (project)\n\n\nCMake Error at /home/burke/.platformio/packages/framework-espidf/tools/cmake/project.cmake:269 (__project):\n  The CMAKE_CXX_COMPILER:\n\n    xtensa-esp32-elf-g++\n\n  is not a full path and was not found in the PATH.\n\n  Tell CMake where to find the compiler by setting either the environment\n  variable \"CXX\" or the CMake cache entry CMAKE_CXX_COMPILER to the full path\n  to the compiler, or to the compiler name if it is in the PATH.\nCall Stack (most recent call first):\n  CMakeLists.txt:3 (project)\n\n\n\n========================= [FAILED] Took 22.89 seconds ========================="

valeros, 2020-09-02:
Hi @mntolia! Looks like corrupted toolchain to me, try to delete folder `/home/burke/.platformio/packages/toolchain-xtensa32` and run your project again.

closed 2020-09-02 by valeros

burke54, 2020-09-02:
valeros,

I tried that many times before opening the problem,  I even removed Visual Studio and reinstalled, (including deleting ALL hidden configuration files).  I then formatted my partition and reinstalled linux (MX Linux 19.2) and got the same results.

The problem was MX Linux by default does not install 'pip'.  This keeps the xtensa tool chain from installing correctly.  Once I got pip installed everything worked as it should.

case closed.