platformio/platformio-home #1171: Libraries: Search

hacker5510, 2020-09-02:
PIO Core Call Error: "'utf-8' codec can't decode byte 0xa8 in position 1567: invalid start byte"

valeros, 2020-09-02:
Hi @hacker5510 ! Could you please share some steps to reproduce the issue? What library were you trying to install/search?

hacker5510, 2020-09-02:
@valeros 

1.Click PIO Home Button
2.Click Tag : Esp32
 
[image]

valeros, 2020-09-02:
Unfortunately, I cannot reproduce the issue. Do you have any non-unicode symbols in your username?

hacker5510, 2020-09-02:
@valeros Thanks you.:-)

closed 2020-09-02 by hacker5510

valeros, 2020-09-02:
Have you resolved the issue? Would be great if you could add some comments on your solution for other users.