platformio/platformio-home #1204: Could not load platform updates

Talha909, 2020-09-07:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 393, in platform_update\r\n    and not PlatformFactory.new(pkg).are_outdated_packages()\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\platformio\\platform\\_packages.py\", line 134, in are_outdated_packages\r\n    pkg, self.get_package_spec(pkg.metadata.name)\r\n  File \"c:\\users\\hp\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 48, in outdated\r\n    self.pick_best_registry_version(reg_pkg[\"versions\"]) or {}\r\nKeyError: 'versions'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-09-09:
Could you reproduce this issue?

Talha909, 2020-09-10:
Sure. I have open the visual studio then go into the Platefom.ino extension and click on platforms and then click on update. And i got the above error. I have already update all the things. I also delete the pervious installation and install new one and got the same message.

ivankravets, 2020-09-10:
Could you mail me@ikravets.com to get more details on how to Debug this issue. We are going to release 5.0.1 of PIO Core today and it would be great to fix it. Thanks.

ivankravets, 2020-09-11:
Resolved in https://github.com/platformio/platformio-core/issues/3666

closed 2020-09-11 by ivankravets