platformio/platformio-home #1228: Could not load platform updates

VIXIVIXIV, 2020-09-09:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 393, in platform_update\r\n    and not PlatformFactory.new(pkg).are_outdated_packages()\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\platform\\_packages.py\", line 134, in are_outdated_packages\r\n    pkg, self.get_package_spec(pkg.metadata.name)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 48, in outdated\r\n    self.pick_best_registry_version(reg_pkg[\"versions\"]) or {}\r\nKeyError: 'versions'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-09-09:
Could you reproduce this issue?

Majubs, 2020-09-10:
I'm getting the same error message, started happening after last update. When I run `pio run` it complains about a library I have in my `/lib` folder. Here is the entire log:
> 
> Processing nodemcuv2 (platform: espressif8266; framework: arduino; board: nodemcuv2)
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
> PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
> HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
> PACKAGES:
>  - framework-arduinoespressif8266 3.20704.0 (2.7.4)
>  - tool-esptool 1.413.0 (4.13)
>  - tool-esptoolpy 1.20800.0 (2.8.0)
> 
>  - tool-mklittlefs 1.203.200522 (2.3)
>  - tool-mkspiffs 1.200.0 (2.0)
>  - toolchain-xtensa 2.40802.200502 (4.8.2)
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
>     return self.method(*nargs, **kwargs)
>   File "C:\Users\majubs\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
>     project.install_dependencies()
>   File "C:\Users\majubs\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
>     lm.install(spec)
>   File "c:\users\majubs\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
>     spec, silent=silent, skip_dependencies=skip_dependencies, force=force
>   File "c:\users\majubs\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
>     force=force,
>   File "c:\users\majubs\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 99:
>     pkg = self.install_from_registry(spec, search_filters, silent=silent)
>   File "c:\users\majubs\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 86:
>     version = self.pick_best_registry_version(package["versions"], spec)

If I run `pio update`, I'm able to build normally, but I get the following warning:

`Warning! Could not find the package with 'lib/NTPClient' requirements for your system 'windows_amd64'`

But after that, if I try to build again, it fails again with the same log as above. Everything was building normally before the last PlatformIO update to version 3.3.0.

ivankravets, 2020-09-11:
What is your `platformio.ini`?

Majubs, 2020-09-11:
Here, I'm using Atom btw

```
[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
board_build.filesystem = littlefs

monitor_speed = 115200

build_flags =
  !python print_git_rev.py
  -I src/file_system
  -I src/wireless
  -I src/protocols
  -I src/buffer
  -I src/platform
  -I src/helpers

lib_deps =
  PubSubClient@2.7
  ArduinoJson@6.12.0
  NTPClient=lib/NTPClient
  ESP8266Ping=lib/ESP8266Ping

extra_scripts = upload_fw_info.py
```