platformio/platformio-home #1246: Libraries: Installed

tonydee2377, 2020-09-10:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 307, in lib_list\r\n    items = lm.legacy_get_installed()\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_legacy.py\", line 61, in legacy_get_installed\r\n    return [self.legacy_load_manifest(pkg) for pkg in self.get_installed()]\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 211, in get_installed\r\n    pkg.metadata = self.build_metadata(pkg_dir, spec)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 182, in build_metadata\r\n    spec=spec,\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 337, in __init__\r\n    self.version = version\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 368, in version\r\n    else cast_version_to_semver(value)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\version.py\", line 23, in cast_version_to_semver\r\n    return semantic_version.Version(value)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 105, in __init__\r\n    major, minor, patch, prerelease, build = self.parse(version_string, partial)\r\n  File \"C:\\Users\\Antonio\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 309, in parse\r\n    match = version_re.match(version_string)\r\nTypeError: expected string or bytes-like object\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-09-11:
Do you use the latest PlatformIO IDE 2.0?

tonydee2377, 2020-09-11:
PlatformIO Core 5.0.1  Home 3.3.0

ivankravets, 2020-09-11:
Does it work now?

tonydee2377, 2020-09-11:
No, removed multicore Platformio, now it works, but if I go through the menus, it gives me core errors and gives me the red report button.
I think I should reinstall Platformio to clean this error, everything started a few days after the update. You know the latest updates must never be done, they are tests, but I did them, and now I can't compile my project. I even changed the platformio.ini file

ivankravets, 2020-09-12:
Do you have libraries in `C:\Users\Antonio\.platformio\lib` folder? Please remove this global `lib` folder and install libraries per proejct https://docs.platformio.org/en/latest/librarymanager/quickstart.html

closed 2020-09-12 by ivankravets