platformio/platformio-home #1255: Libraries: Updates

tnn85, 2020-09-11:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n    return f(get_current_context(), *args, **kwargs)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 247, in lib_update\r\n    lm = LibraryPackageManager(storage_dir)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\library.py\", line 30, in __init__\r\n    PackageType.LIBRARY, package_dir or get_project_global_lib_dir()\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 54, in __init__\r\n    self.package_dir = self.ensure_dir_exists(package_dir)\r\n  File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 98, in ensure_dir_exists\r\n    assert os.path.isdir(path)\r\nAssertionError\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

thomastech, 2020-09-12:
+1. After updating Platformio to version 5.0.1 I also experience a similar error message when I attempt library updates.

Other observations:
1. The PIO home's library update screen shows endless spinning of the loading icons.

2. I also see another error, as follows:
`PIO Core Call Error: "Error: Please setup environments in `platformio.ini` file"`
I've made no changes to platformio.ini, this error appeared after updating Platformio.

3. I can still successfully compile. So far the only oddity is that I cannot update a library using the built-in Update function.

Troubleshooting status:
I've rerun platformio upgrade ("pio upgrade" and "pip install -U platformio" commands). They both report I have the latest release.

- Thomas

ivankravets, 2020-09-12:
@thomastech what is your OS? Could provide a simple project to reproduce this issue?

tnn85, 2020-09-12:
Win 10 Pro 64 Bit. 
I'm not sure but could it be an issue with the renaming of the python interpreter? I have also updated my python interpreter from 3.7x to 3.8.6rc1, deleted "penv"  and "python37" folder in .platformio and now it seems to work.

ivankravets, 2020-09-12:
No, you just installed stable PlatformIO Core 5.0.1 that has a fix for this issue.

closed 2020-09-12 by ivankravets

thomastech, 2020-09-12:
FWIW, I am also using Win 10 Pro 64 Bit. This problem affects all my projects. When Platformio's library update function is used all the projects that have a new library are populated in the list and their loading icons spin forever. 

UPDATE: Today there are some new error messages. Some projects now cause this type of error message:
`PIO Core Call Error: "Error: Unknown environment names {snip}"`

I haven't had time to investigate the new error messages. If I can't resolve them I will create a new issue ticket rather than add unrelated issues to this one. But any advice is welcome.

- Thomas

tnn85, 2020-09-15:
@ivankravets I cheered too soon. I still have problems. This time with the Library Update function in PIO Home (Not the Platforms Update function like eariler) . The loading icon is spinning endlessly and i get this error:

Libraries: Updates
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n cli() # pylint: disable=no-value-for-parameter\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n rv = self.invoke(ctx)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n return callback(*args, **kwargs)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 247, in lib_update\r\n lm = LibraryPackageManager(storage_dir)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\library.py\", line 30, in __init__\r\n PackageType.LIBRARY, package_dir or get_project_global_lib_dir()\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 54, in __init__\r\n self.package_dir = self.ensure_dir_exists(package_dir)\r\n File \"C:\\Users\\user1\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 98, in ensure_dir_exists\r\n assert os.path.isdir(path)\r\nAssertionError\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

@thomastech  I can confirm it affects all projects.

ivankravets, 2020-09-15:
Please click on the "Terminal" icon on the blue bottom bar and type `pio upgrade`. Restart VSCode.

tnn85, 2020-09-15:
PS C:\Users\user1\OneDrive\Desktop\CrealityEnder3_3DPrinter\Firmware\Marlin-2.0.x> pio upgrade
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.

thomastech, 2020-09-15:
@tnn85: I'm on 5.0.1 too. Ran into the same problems as you. Fortunately I've been able to eliminate the error messages in the Library Update function. My path to success was by a lucky break.  Here's the story:

While working on a project I noticed that the \.pio\libdeps\ folder was filled with library folders from other projects. I had no idea how they got there. I deleted all the extraneous folders and confirmed my project compiled. So far, so good.

Then I ran the Library update function to see what it would do. The error messages appeared as usual, along with the spinning loading icons. Curious, I checked the project's libdeps folder and it had been repopulated again with those extra folders!

Then I checked the \.pio\libdeps\ folder in all my other Platformio projects. They too had extra lib folders from my other projects. It seems that there's something odd going on with Platformio 5's Library Update, since these mysterious additions did not occur until after the recent upgrade.

My solution:
1. Closed down Platformio.
2. Deleted some Platformio projects that were no longer needed. 
3. Visited each Platformio Project and deleted the folders in \.pio\libdeps\ that did not belong in the project.
4. Launched Platformio. Confirmed that the projects compiled to ensure I didn't cause any harm. 
5. Ran Library Update. Confirmed the error messages and endlessly spinning loading icons were gone.

Platformio seems to be working normally now. I don't know if library update is working correctly (haven't had a new library appear yet), but at least the error messages are resolved.

- Thomas