platformio/platformio-home #1257: Could not install platform

etcwebdesigns, 2020-09-12:
PIO Core Call Error: "Platform Manager: Installing c:/user/babyg/downloads/Seeed_Arduino_LIS3DHTR-master.zip\r\n\n\nError: VCS: Unknown repository type c:/user/babyg/downloads/Seeed_Arduino_LIS3DHTR-master.zip"

etcwebdesigns, 2020-09-12:
I am not sure that I am doing this correctly: I have a Seeed-Arduino 3 Axis accelerometer and the library is the LIS3DHTR but this library is not in the PlatformIO library. So I tried to install it, I wasn't sure how to install it until I ran across this section of custom install. I don't know if this is just for the IDE or not but it seemed to make sense. I even tried to install it using the Git web page for the repository but that didn't work either. Any help would be awesome.

ivankravets, 2020-09-12:
You need to use `lib_deps` https://docs.platformio.org/en/latest/librarymanager/quickstart.html

closed 2020-09-12 by ivankravets