platformio/platformio-home #1258: Libraries: Could not install library

someWhoAnon, 2020-09-12:
PIO Core Call Error: "Library Storage: C:\\Users\\Yoba\\Documents\\PlatformIO\\Projects\\Esp32Box_platformIO\\.pio\\libdeps\\esp32dev\r\nLibrary Manager: Installing me-no-dev/ESP Async WebServer @ ^1.2.3\r\nLibrary Manager: ESP Async WebServer @ 1.2.3 has been installed!\r\nLibrary Manager: Installing dependencies...\r\nLibrary Manager: Installing ESPAsyncTCP\r\nLibrary Manager: ESPAsyncTCP @ 1.2.2 has been installed!\r\nLibrary Manager: AsyncTCP @ 1.1.1 is already installed\r\nLibrary Manager: Installing Hash\r\n\n\nError: Could not find the package with 'Hash' requirements for your system 'windows_amd64'"

ivankravets, 2020-09-13:
Please upgrade PlatformIO Core to the latest version using `pio upgrade` command.

closed 2020-09-13 by ivankravets

wrgallo, 2020-10-07:
> Please upgrade PlatformIO Core to the latest version using `pio upgrade` command.

PIO Core Call Error: "Library Storage: D:\\ControladoraPressaoSuccao\\Firmware\\Firmware\\FirmwareCISA_ESP32\\.pio\\libdeps\\esp32dev\r\nLibrary Manager: Installing me-no-dev/ESP Async WebServer @ ^1.2.3\r\nLibrary Manager: ESP Async WebServer @ 1.2.3 has been installed!\r\nLibrary Manager: Installing dependencies...\r\nLibrary Manager: Installing ESPAsyncTCP\r\nLibrary Manager: ESPAsyncTCP @ 1.2.2 has been installed!\r\nLibrary Manager: AsyncTCP @ 1.1.1 is already installed\r\nLibrary Manager: Installing Hash\r\n\n\nError: Could not find the package with 'Hash' requirements for your system 'windows_amd64'"

Tried the 'pio upgrade' command and got:
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.