platformio/platformio-home #1261: Libraries: Could not install library

JulienHerbiaEra, 2020-09-12:
PIO Core Call Error: "Library Storage: D:\\source\\repos\\ArduinoCode\\ArduinoCode\\.pio\\libdeps\\esp32dev\r\nLibrary Manager: Installing ottowinter/ESPAsyncWebServer-esphome @ ^1.2.7\r\nLibrary Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!\r\nLibrary Manager: Installing dependencies...\r\nLibrary Manager: Installing ESPAsyncTCP-esphome\r\nLibrary Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!\r\nLibrary Manager: Installing AsyncTCP-esphome\r\nLibrary Manager: AsyncTCP-esphome @ 1.1.1 has been installed!\r\nLibrary Manager: Installing Hash\r\n\n\nError: Could not find the package with 'Hash' requirements for your system 'windows_amd64'"

ivankravets, 2020-09-13:
Please upgrade PlatformIO Core to the latest version via `pio upgrade`

closed 2020-09-13 by ivankravets

loly-girl, 2021-01-14:
Also have this issue:

`C:\Users\loly>platformio lib --global install Hash
Library Storage: C:\Users\loly\.platformio\lib
Library Manager: Installing Hash
Error: Could not find the package with 'Hash' requirements for your system 'windows_amd64'`

`In file included from src/esphome/core/component.h:5:0,
                 from src\esphome\components\adc\adc_sensor.h:3,
                 from src\esphome\components\adc\adc_sensor.cpp:1:
C:\Users\loly\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
*** [.pioenvs\phytomodule\src\esphome\components\adc\adc_sensor.cpp.o] Error 1
In file included from src/esphome/core/component.h:5:0,
                 from src\esphome\components\api\api_connection.h:3,
                 from src\esphome\components\api\api_connection.cpp:1:
C:\Users\loly\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
*** [.pioenvs\phytomodule\src\esphome\components\api\api_connection.cpp.o] Error 1
In file included from src/esphome/core/component.h:5:0,
                 from src\esphome\components\api\proto.h:3,
                 from src\esphome\components\api\api_pb2.h:5,
                 from src\esphome\components\api\api_pb2_service.h:5,
                 from src\esphome\components\api\api_pb2_service.cpp:3:
C:\Users\loly\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
In file included from src/esphome/core/component.h:5:0,
                 from src\esphome\components\api\proto.h:3,
                 from src\esphome\components\api\api_pb2.h:5,
                 from src\esphome\components\api\api_pb2.cpp:3:
C:\Users\loly\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
*** [.pioenvs\phytomodule\src\esphome\components\api\api_pb2_service.cpp.o] Error 1
*** [.pioenvs\phytomodule\src\esphome\components\api\api_pb2.cpp.o] Error 1`

ivankravets, 2021-01-14:
Please upgrade PlatformIO Core to the latest version

ed-french, 2021-01-26:
I also have this issue.

I have upgraded the core (as had the previous respondent)!

Upgrading the core didn't solve the issue.

ivankravets, 2021-01-26:
You can't install the `Hash` library, it is already built-in in the framework.

arlenarlenarlen, 2021-04-05:
@ed-french I had the same problem, if you see this [comment in another thread](https://github.com/platformio/platformio-core/issues/3662#issuecomment-723920616), @ivankravets suggests
```
pio platform install espressif8266 --with-all-packages
```
which fixed the issue for me.