platformio/platformio-home #1265: Libraries: Updates

pbriesch, 2020-09-12:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/click/decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/commands/lib/command.py\", line 247, in lib_update\n    lm = LibraryPackageManager(storage_dir)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/library.py\", line 30, in __init__\n    PackageType.LIBRARY, package_dir or get_project_global_lib_dir()\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/base.py\", line 54, in __init__\n    self.package_dir = self.ensure_dir_exists(package_dir)\n  File \"/Users/pbriesch/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/base.py\", line 98, in ensure_dir_exists\n    assert os.path.isdir(path)\nAssertionError\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

way5, 2020-09-12:
I had the same issue. Temporary solved by using external platformio installation: 
`brew install platformio`
do not forget to uncheck "Use built-in PlatformIO Core" in VScode settings, if applicable

ivankravets, 2020-09-13:
The issue has been fixed. Please upgrade to the latest stable PlatformIO Core 5.0. (Left activity bar > PlatformIO > Upgrade PlatformIO Core)

Please turn on back use built-in PIO Core to avoid future issues. Also, please uninstall brew's version of PIO Core.

closed 2020-09-13 by ivankravets