platformio/platformio-home #1275: Could not install library to GlobalStorage

Kabron287, 2020-09-14:
1. Earlier it was not a problem. An **Install** button appeared to the right of the **Release** button and
it was possible to choose where the lib will be located.

Now this button changed to **Add to Project** and no options. 

It is not acceptable for me at all.
How to correct the problem?

2. I got info about available libraries updates. But the **Update** button is blocked. Only Reveal is active.

Thanks in advance,
Vladimir

ivankravets, 2020-09-14:
Could you show a screenshot of "Updates" page?

Why do you need global libraries?

Kabron287, 2020-09-14:
I'm used to doing this...
And note that I am being asked to install this library in a project for a different platform.
[image]

ivankravets, 2020-09-14:
We added "Global libraries" 6 years ago to keep a similar workflow with Arduino IDE. This is the worst that people can architect. As a result, a ton of reports by developers due to conflicts between multiple projects.

We don't more follow Arduino IDE's workflow.  "CLI with -g option works", yes, you can use it. We will keep CLI compatible with our previous versions.

P.S: Why do you do this work manually? PlatformIO does this automatically for over 1,000,000 devs who use https://docs.platformio.org/en/latest/librarymanager/quickstart.html

closed 2020-09-14 by ivankravets

Kabron287, 2020-09-14:
I don't quite understand what kind of automation we are talking about. Before I added the global libraries, there was no way I could build the project WioWiFiAnalyzer at all, because absolutely idiotic errors appeared permanently one by one.
Also, platformio could not "automatically" find some libs at all. E.g. Arduino_GFX,  Seeed_Arduino_atUnified-1.0.0, Seeed_Arduino_atWiFi-1.0.0, esp-at-lib-develop, etc.

ivankravets, 2020-09-14:
Could you provide a simple project to reproduce this issue where `lib_deps` does not work?

Kabron287, 2020-09-14:
I do not use **lib_deps**.

Kabron287, 2020-09-26:
Well, Ivan, you've convinced me.
I just ran into a very contagious libы from Seeed.
But the question is: usually I worked with folders and never created any projects.
Also, when I get code from Git, it placed in a folder.
If I need to add library in platformio it asced for which project?
I.e. I need to transform folder to project.
What is the way?

Kabron287, 2020-09-26:
Ok I find