platformio/platformio-home #1283: Using lib_extra_dirs appears to create multiple entries in Installed list in Libraries

elguiri, 2020-09-15:
Please refer to forum post... https://community.platformio.org/t/using-lib-extra-dirs-appears-to-create-multiple-entries-in-installed-list-in-libraries/16004

Thanks very much!

OmarAlamoudi, 2021-02-04:
I would like to bring attention to this issue again. Please see the image here: https://community.platformio.org/t/using-lib-extra-dirs-appears-to-create-multiple-entries-in-installed-list-in-libraries/16004/6?u=omaralamoudi showing the same issue occurring with me.

Thanks

ivankravets, 2021-03-09:
Could you provide `platformio.ini` to reproduce this issue?

OmarAlamoudi, 2021-03-18:
Here you go @ivankravets 
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:due]
platform = atmelsam
board = due
framework = arduino
lib_extra_dirs = /Users/omaralamoudi/Dropbox/Arduino/libraries

[platformio]
description = Arduino Due (Programming Port)
```

At this exact moment, I'm encountering this issue: https://community.platformio.org/t/platformio-toolbar-and-home-icon-not-showing-in-vscode/12334

So I'm unable to replicate it, but it was a persistent problem

Gixy31, 2022-12-09:
I confirm that each time you add/work on a project with lib_extra_dir, there is one corresponding entry i nthe installed libraries list! At the end, it's nothing but confusing...