platformio/platformio-home #1297: Erase flash option missed on "Platformio tasks" after update

pabloVecchio, 2020-09-17:
I couldn't find 'Erase flash' option after update platformio. I've got Core 5.0.1 and Home 3.3.0 versions. It's has been removed? It was a very helpful tool

ivankravets, 2020-09-17:
https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks

closed 2020-09-17 by ivankravets

peacepenguin, 2020-11-06:
For anyone else having this issue with missing "erase flash" command in the gui. Just select your specific board from the "Project Environment Switcher" at the bottom of vscode. Once you choose your specific board, the "Erase Flash" option comes back under the env: tree in the gui for your now chosen environment.

https://docs.platformio.org/en/latest/_images/platformio-ide-vscode-task-explorer.png

Smitty357, 2020-11-22:
Still no luck for me. Any other suggestions?

peacepenguin, 2020-11-22:
What platform are you using?

Im using an esp32dev platform and this procedure has worked well for me on many of my systems that were missing the erase flash command.

But I'm not sure what is normal for other platforms.