platformio/platformio-home #1309: Libraries: Updates

DAKarp, 2020-09-18:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py\", line 109, in main\n    cli()  # pylint: disable=no-value-for-parameter\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 829, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 782, in main\n    rv = self.invoke(ctx)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1259, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 1066, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/core.py\", line 610, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/click/decorators.py\", line 21, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/commands/lib/command.py\", line 247, in lib_update\n    lm = LibraryPackageManager(storage_dir)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/library.py\", line 29, in __init__\n    super(LibraryPackageManager, self).__init__(\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py\", line 54, in __init__\n    self.package_dir = self.ensure_dir_exists(package_dir)\n  File \"/Users/daaron/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py\", line 98, in ensure_dir_exists\n    assert os.path.isdir(path)\nAssertionError\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2020-09-19:
Please open VSCode > Left Side Bar > PlatformIO (Icon) > Quik Access > PlatformIO Core CLI and type 
```
pio upgrade --dev
```

Restart VSCode. Does it work now?

[image]

closed 2020-09-19 by ivankravets

DAKarp, 2020-09-27:
Hey Ivan - tried this, and now I'm getting results in the Libraries -> Updates view, but all of the Update buttons are grayed out. Thoughts?

DAKarp, 2020-09-27:
Also, the install button is now missing from all library pages. I tried uninstalling and reinstalling the PlatformIO extension in VSCode, but it made no difference.

DAKarp, 2020-09-27:
Ok, I think I've fixed *part of* it. In case anyone else if having this trouble, here's what I did:

1. Return to the PlatformIO Core CLI (see above)
2. Type "pio update"
3. It was able to update a bunch of libraries. While doing so, I saw that it said [Detached] next to "Espressif 32."
4. In the UI, I went to Platforms -> Installed.
5. I located Espressif 32, then uninstalled and reinstalled.

This fixed the library update issue. But new libraries are all still missing the Install button, so I can't install any new libraries. If anyone knows what else I can do, I'd appreciate it.

DAKarp, 2020-09-27:
Ok, finally found out why there's no install button:

https://github.com/platformio/platformio-home/issues/1275

Pretty upset about this.