platformio/platformio-home #1317: Could not load platform data

vollandg, 2020-09-19:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 348, in platform_uninstall\r\n    if pm.uninstall(platform):\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\platform.py\", line 76, in uninstall\r\n    pkg = self.get_package(spec)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 232, in get_package\r\n    spec = self.ensure_spec(spec)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 103, in ensure_spec\r\n    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 119, in __init__\r\n    self._parse(raw)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 222, in _parse\r\n    raw = parser(raw)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 245, in _parse_requirements\r\n    self.requirements = tokens[1].strip()\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\meta.py\", line 162, in requirements\r\n    else semantic_version.SimpleSpec(str(value))\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 618, in __init__\r\n    self.clause = self._parse_to_clause(expression)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 1014, in _parse_to_clause\r\n    return cls.Parser.parse(expression)\r\n  File \"C:\\Users\\darek\\.platformio\\penv\\lib\\site-packages\\semantic_version\\base.py\", line 1034, in parse\r\n    raise ValueError(\"Invalid simple block %r\" % block)\r\nValueError: Invalid simple block 'src-de6b279104eee6e886c6740cfbf2debe'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-09-19:
It was fixed in the latest PIO Core. Please click on "terminal" icon on the blue bar of VSCode and type `pio upgrade`. Restart VSCode.

vollandg, 2020-09-19:
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.

And the same errors.. 

Thanks 😊
Od: Ivan Kravets
Wysłano: sobota, 19 września 2020 17:17
Do: platformio/platformio-home
DW: vollandg; Author
Temat: Re: [platformio/platformio-home] Could not load platform data (#1317)

It was fixed in the latest PIO Core. Please click on "terminal" icon on the blue bar of VSCode and type pio upgrade. Restart VSCode.

ivankravets, 2020-09-19:
Could you archive `C:\Users\darek\.platformio\platforms` folder and share on Dropbox or another service?

closed 2020-11-25 by stale[bot]