platformio/platformio-home #1346: Could not update platform

haplopeart, 2020-09-23:
PIO Core Call Error: "Platform C:\\Users\\haplo\\.platformio\\platforms\\nxplpc-arduino-lpc176x\r\n--------\r\nUpdating https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip 0.1.1                              [Up-to-date]\r\nUpdating platformio/toolchain-gccarmnoneeabi  1.90301.200702 @ >=1.60301.0       [Up-to-date]\r\nUpdating platformio/framework-arduino-lpc176x 0.2.4 @ >=0.2.2                    [Outdated 0.2.5]\r\nTool Manager: Installing platformio/framework-arduino-lpc176x @ 0.2.5\r\n\n\nError: Traceback (most recent call last):\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 421, in platform_update\r\n    platform, only_packages=only_packages, only_check=only_check, silent=silent\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\platform.py\", line 120, in update\r\n    p.update_packages(only_check)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\platform\\_packages.py\", line 128, in update_packages\r\n    show_incompatible=False,\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 114, in update\r\n    return self._update(pkg, outdated, silent=silent)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 156, in _update\r\n    silent=silent,\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_install.py\", line 49, in install\r\n    spec, silent=silent, skip_dependencies=skip_dependencies, force=force\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_install.py\", line 99, in _install\r\n    pkg = self.install_from_registry(spec, search_filters, silent=silent)\r\n  File \"C:\\Users\\haplo\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_registry.py\", line 86, in install_from_registry\r\n    version = self.pick_best_registry_version(package[\"versions\"], spec)\r\nKeyError: 'versions'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

haplopeart, 2020-09-24:
I uninstalled the previous version during complile the new version was downlaoded and successfuly updated.

ivankravets, 2020-10-30:
Please open PlatformIO CLI and type `pio upgrade`. Restart VSCode. Is fixed in the latest PlatformIO Core 5.0.2

closed 2020-10-30 by ivankravets